TerosTechnology/vscode-terosHDL #614: Error schematic generation if top units more than one on VHDL files

ila-embsys, 2024-05-20:
## Describing the bug
By pressing the 'Schematic Viewer' button on the 'Dependency Tree' panel, the Schematic Viewer window shows a blank page.

In the same time, the Output page for 'TerosHDL: Global' shows the log like that:
```
[info] yosys -m ghdl -p "ghdl --std=08 -fsynopsys a.vhd b.vhd tb_1.vhd tb_2.vhd -e; hierarchy -top tb_1; proc;  ; write_json teroshdl_yosys_output.json; stat"
[info] Yosys 0.33
-- Running command
1. Executing GHDL.
[info] error: no top unit found
ERROR: vhdl import failed.
[error] Yosys failed
```

The project, that produce error, contains two source files: `tb_1.vhd` and `tb_2.vhd`. Any of them could be set as a top file by pressing 'Select source as toplevel' on 'Sources' panel. That choosing changes the list of 'Dependency tree' and the list of 'Runs' panel. Also, that choosing reflects the '-top' argument value that passing to GHDL inside Yosys:
 > 'ghdl --std=08 -fsynopsys a.vhd b.vhd tb_1.vhd tb_2.vhd -e; hierarchy **-top tb_1**; proc;  ; write_json teroshdl_yosys_output.json; stat'

However, the command still produces error.

## Investigating
To find where the error comes from, that generated command could be passed to Yosys CLI directly.

Run Yosys:

`yosys -m ghdl`

Then execute first part of generated command:

`yosys> ghdl --std=08 -fsynopsys a.vhd b.vhd tb_1.vhd tb_2.vhd -e;`

The error exactly here:

```
1. Executing GHDL.
error: no top unit found
ERROR: vhdl import failed.
```

### GHDL internals
This error message `no top unit found` probably comes from [here](https://github.com/ghdl/ghdl/blob/dc17919eb3d401da6843cb4e298b9005d36d93e9/src/ghdldrv/ghdllocal.adb#L1884).

That `Find_Top_Entity` function, probably, [here](https://github.com/ghdl/ghdl/blob/dc17919eb3d401da6843cb4e298b9005d36d93e9/src/vhdl/vhdl-configuration.adb#L1131-L1151).

One of the conditions that the function could return nothing is [top level files more than one](https://github.com/ghdl/ghdl/blob/dc17919eb3d401da6843cb4e298b9005d36d93e9/src/vhdl/vhdl-configuration.adb#L1141-L1145):

```ada
if Top.Nbr_Top_Entities = 1 then
   return Top.First_Top_Entity;
else
   return Null_Iir;
end if;
```

### Resolution
By trying to remove one of the `tb` files from the project 'sources' list, to left only one top file, the error disappears. 🎉
With the ability to choose any top file in the 'sources' list, not obvious that 'Schematic viewer' will not work if top file more then one.

I'm not sure my workflow is correct when I switch project's top file and run different 'Runs', but I tried and failed with 'Schematic Viewer'.

## Possible resolve

An empty '-e' argument on the generated command "ghdl --std=08 -fsynopsys a.vhd b.vhd tb_1.vhd tb_2.vhd **-e**; hierarchy -top tb_1;" could be filled with the same value as for the flag '-top' on "hierarchy **-top tb_1**;". With that argument passing, that whole generated command works fine even with the multiple top files.

That '-e' probably comes from [this line](https://github.com/TerosTechnology/vscode-terosHDL/blob/0502d24e0b169c8c7b1afcf589b0daab1e68acb6/packages/teroshdl/src/features/yosys.ts#L77), but I don't know how it should work.

___
**Environment information:**
 - OS: Ubuntu 24.04 LTS
 - VSCode version: 1.89.1
 - TerosHDL: v5.0.12
 - Yosys: 0.33
 - GHDL: 4.1.0
 - Schematic viewer backend: GHDL (module) + Yosys

ila-embsys, 2024-05-20:
Pretty sure the reasons of https://github.com/TerosTechnology/vscode-terosHDL/issues/381 is the same.

ila-embsys, 2024-05-23:
This [line](https://github.com/TerosTechnology/vscode-terosHDL/commit/d5dd14a47b1db9419a140a30d27bee5d3048076a#diff-deb1d257281f241682d267c3598d226fd1ba9ed278eb903f782642646dee8c1cR135-R137) which added for the upcoming release probably will fix the issue.

qarlosalberto, 2025-03-18:
Was the issue fixed? @ila-embsys

ila-embsys, 2025-03-19:
I checked the next release after v5.0.12 where the issue was noticed: v6.0.4 passes '-e' flag correctly with the actually selected top module.

Also, I checked the last version of v6 branch: v6.0.14 have the same correct behavior.

Looks like the issue is resolved. At least for v6 branch.