TerosTechnology/vscode-terosHDL #624: Schematic viewer dosen't work

deleteTh, 2024-06-19:
TerosHDL:Global output: 2024-06-19 09:53:02.780 [info] [00000.001204] ERROR: Can't guess frontend for input file `d:/workSpace/sim/fifo/axis_async_fifo.v;' (missing -f option)!

Configure environment:
 - OS: win10
 - VSCode version:1.90.0
 - teroshdl version:v6.0.1
 - python version:3.12.0rc3
 - 
1.  "yowasp-yosys --help" can in CMD and Power Shell work.

2.  In CMD window use "make --version" output:
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-pc-mingw32

3. My environment variable configure:
D:\devSoft\Python\Python312\Scripts\
D:\devSoft\Python\Python312\
C:\Windows\System32

4. In schematic viewer: General >> Select the backend >> YoWASP(Only Verilog/SV), other configurations remain by default.

qarlosalberto, 2024-06-19:
Can you share an example to reproduce it?

deleteTh, 2024-06-19:
[axis_async_fifo.txt](https://github.com/user-attachments/files/15896223/axis_async_fifo.txt)
[image]

deleteTh, 2024-06-19:
I found that many verilog file cannot be generated.

deleteTh, 2024-06-19:
&gt; TerosHDL:Global output: 2024-06-19 09:53:02.780 [info] [00000.001204] 
 &gt; ERROR: Can't guess frontend for input file 
 &gt; `d:/workSpace/sim/fifo/axis_async_fifo.v;' (missing -f option)! 
 &gt; 
 &gt; Configure environment: 
 &gt; 
 &gt;    - OS: win10 
 &gt;    - VSCode version:1.90.0 
 &gt;    - teroshdl version:v6.0.1 
 &gt;    - python version:3.12.0rc3 
 &gt;    - 
 &gt; 
 &gt; 
 &gt;    1. 
 &gt; 
 &gt;    "yowasp-yosys --help" can in CMD and Power Shell work. 
 &gt;    2. 
 &gt; 
 &gt;    In CMD window use "make --version" output: 
 &gt;    GNU Make 3.81 
 &gt;    Copyright (C) 2006 Free Software Foundation, Inc. 
 &gt;    This is free software; see the source for copying conditions. 
 &gt;    There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A 
 &gt;    PARTICULAR PURPOSE. 
 &gt; 
 &gt; This program built for i386-pc-mingw32 
 &gt; 
 &gt;    3. 
 &gt; 
 &gt;    My environment variable configure: 
 &gt;    D:\devSoft\Python\Python312\Scripts 
 &gt;    D:\devSoft\Python\Python312 
 &gt;    C:\Windows\System32 
 &gt;    4. 
 &gt; 
 &gt;    In schematic viewer: General &gt;&gt; Select the backend &gt;&gt; YoWASP(Only 
 &gt;    Verilog/SV), other configurations remain by default. 
 &gt; 
 &gt; — 
 &gt; <https://github.com/TerosTechnology/vscode-terosHDL/issues/624&gt;, or 
 &gt; unsubscribe 
 &gt; . 
 &gt; ID: ***@***.***&gt; 
 &gt;

qarlosalberto, 2024-06-19:
Could you try with this version? https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
I have added more log. Yo should see on top the current command:

[image]

Share it here, so I can see the problem.

deleteTh, 2024-06-20:
[image]

deleteTh, 2024-06-20:
[image]
[image]
[ErrorInfo.txt](https://github.com/user-attachments/files/15909257/ErrorInfo.txt)

deleteTh, 2024-06-20:
When I use this command : 
yowasp-yosys -p 'read_verilog -sv d:/workSpace/sim/fifo/axis_async_fifo.v; ; hierarchy -top axis_async_fifo; proc; ; write_json C:/Users/LBWB_WT/.teroshdl_0hHab; stat', then CMD window was no error.
[image]

AKA-DongDong, 2024-09-23:
> When I use this command : yowasp-yosys -p 'read_verilog -sv d:/workSpace/sim/fifo/axis_async_fifo.v; ; hierarchy -top axis_async_fifo; proc; ; write_json C:/Users/LBWB_WT/.teroshdl_0hHab; stat', then CMD window was no error. ![5](https://private-user-images.githubusercontent.com/122022235/341302090-b5af2ae1-9913-49be-9178-c60d144deeee.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwNzIwNTksIm5iZiI6MTcyNzA3MTc1OSwicGF0aCI6Ii8xMjIwMjIyMzUvMzQxMzAyMDkwLWI1YWYyYWUxLTk5MTMtNDliZS05MTc4LWM2MGQxNDRkZWVlZS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyM1QwNjA5MTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jOTJkMTExZmM4MDNiOTExYjFiM2E1MDg4NGJhNDA0ZTA2MzQ1NTQ4ZWQ3ZGQzODZmZTUyMmUwMDZmNDYyYzcwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.VI0f2Pex-uqzYrikSxJhlqYkhSeuT1g0MHYHmmRvX-c)

junlzu, 2024-09-29:
TerosHDL: v6.0.7
Python 3.11.1
[image]

qarlosalberto, 2024-09-29:
Try with the new version in the market `v6.0.9`.

junlzu, 2024-09-30:
> Try with the new version in the market `v6.0.9`.

Thank you!

TerosHDL: v6.0.7
The following informations is showed. 
2024-09-30 14:06:21.039 [info] Optimizing module f_adder.
Optimizing module half_add.

2024-09-30 14:06:21.039 [info] ERROR: Can't open output file `C:\Users\Jun\.teroshdl_4rjJX' for writing: Operation not permitted

2024-09-30 14:06:21.105 [error] Yosys failed.

The same result is showed when I executed vs by administrator or user.  everyone can write file in 'C:\Users\Jun\'.

qarlosalberto, 2024-09-30:
Are you using yowasp-yosys or yosys in your config?

qarlosalberto, 2024-09-30:
Can you write in `C:\Users` ?

junlzu, 2024-09-30:
> Are you using yowasp-yosys or yosys in your config?

YoWASP
[image]

junlzu, 2024-09-30:
> Are you using yowasp-yosys or yosys in your config?

YoWASP

junlzu, 2024-09-30:
> Can you write in `C:\Users` ?

No, I can't create any file in 'C:\Users', the dialog is showed when I create a folder in 'C:\Users'. 
[image]

But I can create and write a new file in 'C:\Users\Jun'.

qarlosalberto, 2024-09-30:
Try to define the environment system variable `USERPROFILE` to: `C:\Users\Jun`

qarlosalberto, 2024-09-30:
Aso, update ` yowasp-yosys`

junlzu, 2024-09-30:
> Aso, update ` yowasp-yosys`

Step1:
[image]

Step2:
pip install --upgrade yowasp-yosys

Installing collected packages: yowasp-yosys
  Attempting uninstall: yowasp-yosys
    Found existing installation: yowasp-yosys 0.35.0.0.post605
    Uninstalling yowasp-yosys-0.35.0.0.post605:
      Successfully uninstalled yowasp-yosys-0.35.0.0.post605
Successfully installed yowasp-yosys-0.45.0.0.post775

but, still report the error.
:(

qarlosalberto, 2024-09-30:
Can you share a screenshot?

qarlosalberto, 2024-09-30:
Has you modified `USERPROFILE` to `C:\Users\Jun` or it was yet that?

The main problem is here:

[image]

It should be `C:\Users\Jun\.teroshdl_4rjJX` and not `C:\Users\Jun.teroshdl_4rjJX`. In trying to found why this happen.

junlzu, 2024-09-30:
[cid:1aafad14-b786-43df-8ae9-c2991a257f31]

imagen.png (view on web)<https://github.com/user-attachments/assets/33e284c4-fc89-4ea3-ba7b-4202644d5506>

junlzu, 2024-09-30:
I executed the command in CMD by administrator. 

yowasp-yosys -p "read_verilog -sv "d:/BaiduSyncdisk/StudyProj/vscode/Verilog/add/f_adder.v"; hierarchy -top f_adder; proc; write_json d:\tmp\.teroshdl_gHxws; stat"
[execinfos.txt](https://github.com/user-attachments/files/17191416/execinfos.txt)

qarlosalberto, 2024-09-30:
Try to run the same command from the folder `d:\tmp`

junlzu, 2024-09-30:
> Try to run the same command from the folder `d:\tmp`
The command was successful
yowasp-yosys -p "read_verilog -sv "d:/BaiduSyncdisk/StudyProj/vscode/Verilog/add/f_adder.v"; hierarchy -top f_adder; proc; write_json .teroshdl_gHxws; stat"
[execinfos2.txt](https://github.com/user-attachments/files/17191938/execinfos2.txt)

junlzu, 2024-09-30:
> Try to run the same command from the folder `d:\tmp`
The command was successful too!
yowasp-yosys -p "read_verilog -sv "d:/BaiduSyncdisk/StudyProj/vscode/Verilog/add/f_adder.v"; hierarchy -top f_adder; proc; write_json d:/tmp/.teroshdl_gHxws; stat"

d:/tmp/.teroshdl_gHxws  is right! not d:\tmp\.teroshdl_gHxws
:)

junlzu, 2024-09-30:
The command was successful too!
yowasp-yosys -p "read_verilog -sv "d:/BaiduSyncdisk/StudyProj/vscode/Verilog/add/f_adder.v"; hierarchy -top f_adder; proc; write_json d:/tmp/.teroshdl_gHxws; stat"

d:/tmp/.teroshdl_gHxws is right! not d:\tmp\.teroshdl_gHxws
:)

junlzu, 2024-09-30:
> Try to run the same command from the folder `d:\tmp`

The command was successful too!
yowasp-yosys -p "read_verilog -sv "d:/BaiduSyncdisk/StudyProj/vscode/Verilog/add/f_adder.v"; hierarchy -top f_adder; proc; write_json d:/tmp/.teroshdl_gHxws; stat"

d:/tmp/.teroshdl_gHxws is right! not d:\tmp\ .teroshdl_gHxws
:)

qarlosalberto, 2024-10-03:
It should be fixed in the market  version.

junlzu, 2024-10-04:
> It should be fixed in the market version.

Thank you! It has been resolved.

AKA-DongDong, 2024-10-10:
> v6.0.4 version, I just find the same error like yours. What should i do？ v5.0.12 is fine. `2024-09-23 14:26:43.099 [info] Executing: C:\WINDOWS\system32\cmd.exe C:\WINDOWS\system32\cmd.exe /d /s /c " yowasp-yosys -p 'read_verilog -sv "d:/rtl_verilog/src/Tx.v"; ; hierarchy -top TxRsp_Ch; proc; hierarchy -check; opt; memory; opt; fsm;; write_json C:\Users\c.teroshdl_soC2d; stat'" 2024-09-23 14:26:43.693 [info] [00000.000000]
> 
> 2024-09-23 14:26:43.695 [info] [00000.002457] ERROR: Can't guess frontend for input file `d:/rtl_verilog/src/Tx.v;' (missing -f option)!`
> 
> > When I use this command : yowasp-yosys -p 'read_verilog -sv d:/workSpace/sim/fifo/axis_async_fifo.v; ; hierarchy -top axis_async_fifo; proc; ; write_json C:/Users/LBWB_WT/.teroshdl_0hHab; stat', then CMD window was no error. ![5](https://private-user-images.githubusercontent.com/122022235/341302090-b5af2ae1-9913-49be-9178-c60d144deeee.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MjcwNzIwNTksIm5iZiI6MTcyNzA3MTc1OSwicGF0aCI6Ii8xMjIwMjIyMzUvMzQxMzAyMDkwLWI1YWYyYWUxLTk5MTMtNDliZS05MTc4LWM2MGQxNDRkZWVlZS5QTkc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwOTIzJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDkyM1QwNjA5MTlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jOTJkMTExZmM4MDNiOTExYjFiM2E1MDg4NGJhNDA0ZTA2MzQ1NTQ4ZWQ3ZGQzODZmZTUyMmUwMDZmNDYyYzcwJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.VI0f2Pex-uqzYrikSxJhlqYkhSeuT1g0MHYHmmRvX-c)

I just found it was fixed in v6.0.10 . Thank you!

closed 2024-10-10 by qarlosalberto