TerosTechnology/vscode-terosHDL #626: SystemVerilog package import support

gtaylormb, 2024-06-19:
The ability to import packages is an important SystemVerilog feature. I use packages in all my designs. The import syntax seems to be understood but it is reported that it can't find the package, but it's in my project. Subsequently, any parameters/functions/symbols defined in my packages show up as errors (undefined variables). Can you please support this feature?

seb-slowik, 2024-09-06:
This is a big one for anyone writing SystemVerilog code. I'm personally waiting for this feature to be supported so, that I could make the transition from using Sublime IDE for UVM to VS Code.

qarlosalberto, 2024-09-06:
> This is a big one for anyone writing SystemVerilog code. I'm personally waiting for this feature to be supported so, that I could make the transition from using Sublime IDE for UVM to VS Code.

I know it, but this is a hard feature. If you need commercial support feel free to send me an email.