TerosTechnology/vscode-terosHDL #628: Schematic generation successfulness depends on project file order if libraries are used

ila-embsys, 2024-06-19:
**Bug description**
An order of files in a project affects a command line to invoke yosys. The different order is the different result.

**To Reproduce**
Let's assume a project that have a `mylib` library.

The project structure will be like that:
```
> mylib
  > mypkg.vhd
> top.vhd
```

If `top.vhd` was added after `mypkg.vhd`, the schematic generation would be fine.

```
-- Running command `ghdl --std=08 -fsynopsys  --work=mylib /workspaces/example/mylib/mypkg.vhd  /workspaces/example/top.vhd  -e top; hierarchy -top top; proc; ; write_json /home/example/.teroshdl_keiPV; stat' --
```

If `top.vhd` was added before `mypkg.vhd`,  the schematic generation would be producing error:

```
-- Running command `ghdl --std=08 -fsynopsys   /workspaces/example/top.vhd --work=mylib /workspaces/example/mylib/mypkg.vhd  -e top; hierarchy -top top; proc; ; write_json /home/vscode/.teroshdl_kRCQT; stat' --

1. Executing GHDL.

[info] error: cannot find entity or configuration top
ERROR: vhdl import failed.

[error] Yosys failed.
```

**Clue of issue**
The difference is the first part of the command line that passed to yosys:
`ghdl --std=08 -fsynopsys  --work=mylib /workspaces/example/mylib/mypkg.vhd  /workspaces/example/top.vhd  -e top;`
vs
`ghdl --std=08 -fsynopsys   /workspaces/example/top.vhd --work=mylib /workspaces/example/mylib/mypkg.vhd  -e top;`

**Code that was used to fill the project files**

`top.vhd`:
```vhdl
library mylib;
use mylib.mypkg;

entity top is
  port (
    a : in integer;
    b : in integer;
    c : out integer
  );
end entity top;

architecture rtl of top is
begin
  c <= mypkg.add(a, b);
end;
```

`mypkg.vhd`
```vhdl
package mypkg is

  function add(a : integer; b : integer) return integer;

end;

package body mypkg is
  function add(a : integer; b : integer) return integer is
  begin
    return a + b;
  end;
end package body;
```

**Environment information**
- OS: Ubuntu 24.04 LTS
- VSCode version: 1.90.1
- TerosHDL: v6.0.1-dev-1ad865aa
- Yosys: 0.33
- GHDL: 4.1.0
- Schematic viewer backend: GHDL (module) + Yosys

**Additional context**
The issue could be hotfixed by adding explicitly before each project file a related library names like `--work=work` and`--work=mylib` and switch back to `--work=work` at the end. Like that:

```
ghdl --std=08 -fsynopsys --work=work /workspaces/example/top.vhd --work=mylib /workspaces/example/mylib/mypkg.vhd --work=work -e top;
```

qarlosalberto, 2024-06-19:
Why do we need the last `--work=work`? What happen if the top is in other library, not in work?

ila-embsys, 2024-06-19:
> Why do we need the last --work=work? 

I guess yosys will search the `-e top` in the last library passed by `--work`.

```
ghdl --std=08 -fsynopsys --work=work  /workspaces/example/top.vhd --work=mylib /workspaces/example/mylib/mypkg.vhd  -e top
1. Executing GHDL.
error: cannot find entity or configuration top
```

> What happen if the top is in other library, not in work?

Didn't though about that case, but I checked that the last `--work` could be set to a value of library where needed to search the desired top.

For example, in the case where the top in the library with name `top`. Like that:
```
> mylib
  > mypkg.vhd
> top
  > top.vhd
```

The command line for yosys could be like that:
```
ghdl --std=08 -fsynopsys  --work=top /workspaces/example/top.vhd --work=mylib /workspaces/example/mylib/mypkg.vhd --work=top -e top;
```

Another variant to set the module for top is to name the top with a library prefix, eg `top.top`. Like that:

```
ghdl --std=08 -fsynopsys  --work=top /workspaces/example/top.vhd --work=mylib /workspaces/example/mylib/mypkg.vhd -e top.top;
```

But this approach for some reason doesn't work if the top module was without a library.

qarlosalberto, 2024-07-03:
Thank you very much! This PR fixes partially: https://github.com/TerosTechnology/vscode-terosHDL/pull/633

I have pending to pass the toplevel library. So I will maintain this issue open.