TerosTechnology/vscode-terosHDL #636: VHDL 2008 Block Comment End Characters Appear in Documentation

jchang-endiag, 2024-07-23:
**Describe the bug**
The VHDL 2008 block comment end characters */ show up in the Module documentation.

**To Reproduce**
1. Open block_comment.vhd in TerosHDL
2. Click _Module documentation preview_.
3. Observe */ appear in _Module documentation_.

**Code**
`/*! my comment */`

**Please complete the following information:**
 - OS: Windows 10
 - VSCode version version 1.91.1

**Screenshots**
[image]

**Additional context**
n/a
[block_comment.zip](https://github.com/user-attachments/files/16354451/block_comment.zip)

qarlosalberto, 2024-08-27:
Block comments aren't supported.

jchang-endiag, 2024-11-13:
Apologies for the late reply.

How do you mean by "Block comments aren't supported."?  I'm a bit confused because block comments seem to work fine, except for the fact that the end block comment characters appear in the documentation.  I'm actually using block comments for documentation presently.