TerosTechnology/vscode-terosHDL #641: No Dependency tree for VUnit projects

james-ziegler, 2024-08-15:
**Describe the bug**

No dependency tree is created for vunit projects. With run.py set as top level, the tree does not populate.

**To Reproduce**

Create a new project using the vunit example. 

**Code**
Code to reproduce the error.

**Please complete the following information:**
 - OS: Windows 10
 - VSCode version version 1.92.1
 - TerosHDL version 5.0.12
 - 
**Screenshots**
If applicable, add screenshots to help explain your problem. **Share the code as text, not as a screenshots!**

**Additional context**
Add any other context about the problem here.

qarlosalberto, 2024-08-26:
Can you try with this version? https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/v6.0.2beta

qarlosalberto, 2024-08-26:
Try this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/v6.0.2beta

And you need to add the sources from run.py, not your run.py: https://terostechnology.github.io/terosHDLdoc/docs/next/tool_manager/source#adding-source

Add files -> Source -> Load from VUnit run.py

If you want to have automatically update your sources you can add the run.py to Watchers view:

[image]

[image]

james-ziegler, 2024-08-26:
I updated to the prerelease, and am still seeing some strange behavior in a VUint project.

The hierarchy pane is still empty. The files under the various libraries do show.

The testbenches are not being populated from run.py and the testbench.

qarlosalberto, 2024-08-26:
For the hierarchy pane you need to select a top file.

It's a bit (or a lot :/) confusing but currently there are two ways:

- You add run.py as source: you can see an run you tests from "run" pane.
- You add the run.py as watcher: you can see the hierarchy but you can't run your tests from "run" pane.

james-ziegler, 2024-08-26:
OK, I guess that should then be marked as either closed or listed as an enhancement. It would be easier for me to use (and explain to co-workers) if the VUnit simulation runs and the hierarchy would appear together. For now, I will suggest different projects for running simulation and looking at the overall code-base with hierarchy.  It is a bit confusing, but thank you for clarifying.

qarlosalberto, 2024-08-27:
The problem here is that, to reduce development time, I tried to use VUnit by fitting it into a generic project type. However, it would clearly be better to create a "VUnit" project type. This way, it would be easier to handle the specific nuances.