TerosTechnology/vscode-terosHDL #647: Improve VHDL language configuration to include regions

ckuhlmann, 2024-08-27:
It would be great if the `vhdl.configuration.json` could be expanded by the following section
```
{
    // [...]

    "folding": { 
        "markers": {
            "start": "^\\s*\\-\\-\\s*#?region\\b",
            "end": "^\\s*\\-\\-\\s*#?endregion\\b"
        }
    }
}
```

so that 
```
    --#region TheRegionLabel
    [Some Code ...]
    --#endregion TheRegionLabel
```    
can be folded by VSCode and shown in the minimap:

[image]

I think this is a simple change for a great improvement in functionality.

qarlosalberto, 2024-08-27:
Can you do a PR?

qarlosalberto, 2024-08-27:
Also can you share a bigger screenshot?

ckuhlmann, 2024-08-27:
The screenshot is 1:1, this is the minimap next to the scroll bar.
I will prepare a PR

ckuhlmann, 2024-08-27:
PR:
[https://github.com/TerosTechnology/vscode-terosHDL/pull/648](https://github.com/TerosTechnology/vscode-terosHDL/pull/648)

closed 2024-09-04 by ckuhlmann