TerosTechnology/vscode-terosHDL #654: Update/Upload TerosHDL on OpenVSX

gfcwfzkm, 2024-09-05:
**Is your feature request related to a problem? Please describe.**
It would be great if TerosHDL further embraces OSS by offering the plugin on the free OpenVSX store.
While there is an plugin, it hasn't been updated in over a year, left stuck at version 2.0.13. This would, in return, make VSCode alternatives like VSCodium or TheiaIDE more attractive. 

**Describe the solution you'd like**
Automate the upload to OpenVSX with every release

2papyrikoo, 2024-09-25:
Hi, I'm using teroshdl for syntax highlighting and has been really handy. I had it configured with yosys and ghdl. However, this update changed all my configuration and apparently there is a new way to check for schematic viewer. I'm having trouble setting it to the new version. Any help?

qarlosalberto, 2024-09-26:
@2papyrikoo are you using VHDL or Verilog in the schematic? Show me output in the console when you open the schematic viewer.

2papyrikoo, 2024-09-26:
I downloaded oss-cad-suite and added the path in tools->yosys
->installation path as explained in the documentation. However, when i
select the standalonr option, it pops a message that "standalone backend is
not supported in this version". So, i followed the other option which was
using yowasp-yosys, and it displayed this in the output " can't guess
frontend for input file
'c:/Users/jpove/OneDrive/Documents/PROYS/VHDL/12C/12Cslave.v;' (missing -f
option)!" And "yosys failed. If i try with a vhdl code, (change it to ghdl
-yosys plugin with oss-cad-suite) it shows yosys failed.

2papyrikoo, 2024-09-26:
> Can you share the console output for the VHDL file?

Sure, here is the output "2024-09-26 16:12:58.773 [info] Executing:
C:\WINDOWS\system32\cmd.exe  C:\WINDOWS\system32\cmd.exe /d /s /c "
C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -m ghdl -p 'ghdl --std=08
-fsynopsys  --work=work
c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\main\src\main.vhd
--work=work -e elevator; hierarchy -top elevator; proc; ; write_json
C:\Users\jpove\.teroshdl_2GUX3; stat'"
2024-09-26 16:12:59.165 [error] Yosys failed."

qarlosalberto, 2024-09-26:
Try to run that command in the terminal manually:
C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -m ghdl -p 'ghdl --std=08
-fsynopsys --work=work
c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\main\src\main.vhd
--work=work -e elevator; hierarchy -top elevator; proc; ; write_json
C:\Users\jpove\.teroshdl_2GUX3; stat

2papyrikoo, 2024-09-26:
It shows an error window with the text "the procedure entry point
_ZSt21ios_base_library_initv could not be located in the dynamic link
library C:\OSs-CAD-SUITE\oss-cad-suite\bin\yosys.exe"

2papyrikoo, 2024-09-26:
If i try yosys(only Verilog/SV) shows this in the output "2024-09-26
16:24:44.320 [info] Executing: C:\WINDOWS\system32\cmd.exe
C:\WINDOWS\system32\cmd.exe /d /s /c "
C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p 'read_verilog -sv
"c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\I2C\I2Cslave.v"; ; hierarchy
-top I2CslaveWith8bitsIO; proc; ; write_json
C:\Users\jpove\AppData\Local\Temp\f-2024826-5192-87xwap.acpee; stat'"
2024-09-26 16:24:45.010 [error] Yosys failed"

qarlosalberto, 2024-09-26:
Hmm it seems an error in Yosys side. Can you try another version of OSS CAD?

qarlosalberto, 2024-09-26:
Try to change your OSS CAD version.

2papyrikoo, 2024-09-26:
I would, but there seems to be only one version for windows which was the
one i installed of course

2papyrikoo, 2024-09-27:
It seems that if i run this command (which is almost the same) it works i think but on the command window. C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p "read_verilog -sv \"c:/Users/jpove/OneDrive/Documentos/PROYS/VHDL/I2C/I2Cslave.v\"; hierarchy -top I2CslaveWith8bitsIO; proc; write_json C:/Users/jpove/AppData/Local/Temp/f-2024826-5192-87xwap.acpee; stat"

qarlosalberto, 2024-09-27:
Try the following commands and let me know what of them work fine:

1. `C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p "read_verilog -sv "c:/Users/jpove/OneDrive/Documentos/PROYS/VHDL/I2C/I2Cslave.v"; hierarchy -top I2CslaveWith8bitsIO; proc; write_json C:/Users/jpove/AppData/Local/Temp/f-2024826-5192-87xwap.acpee; stat"`

2. `C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p 'read_verilog -sv
"c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\I2C\I2Cslave.v"; ; hierarchy
-top I2CslaveWith8bitsIO; proc; ; write_json
C:\Users\jpove\AppData\Local\Temp\f-2024826-5192-87xwap.acpee; stat'`

3. `C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p "read_verilog -sv
'c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\I2C\I2Cslave.v'; ; hierarchy
-top I2CslaveWith8bitsIO; proc; ; write_json
C:\Users\jpove\AppData\Local\Temp\f-2024826-5192-87xwap.acpee; stat"`

4. `C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p "read_verilog -sv
c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\I2C\I2Cslave.v; ; hierarchy
-top I2CslaveWith8bitsIO; proc; ; write_json
C:\Users\jpove\AppData\Local\Temp\f-2024826-5192-87xwap.acpee; stat"`

2papyrikoo, 2024-09-27:
Sure, the first command  works normally I guess, it shows the statistics
table.
[image: image.png]
The second command shows an error.
[image: image.png]

The third command cannot acces the file.
[image: image.png]
and the fourth command shows this.
[image: image.png]
which i suppose works fine, as well as the first command.

qarlosalberto, 2024-09-27:
> Sure, the first command works normally I guess, it shows the statistics table. [image: image.png] The second command shows an error. [image: image.png] The third command cannot acces the file. [...]

I have uploaded a new version to the market. Let me know if it's fixed now.

2papyrikoo, 2024-09-27:
2024-09-27 13:58:49.927 [info] Executing: C:\WINDOWS\system32\cmd.exe
 C:\WINDOWS\system32\cmd.exe /d /s /c "
C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p '; proc; write_json
C:\Users\jpove\AppData\Local\Temp\f-2024827-35788-15ar1sw.324f; stat'"
2024-09-27 13:58:50.053 [error] Yosys failed.
I used the yosys(only Verilog/SV) option. However, I tried it on the
command prompt. and appeared this
[OSS CAD Suite]
C:\OSS-CAD-SUITE\oss-cad-suite>C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe
-p '; proc; write_json
C:\Users\jpove\AppData\Local\Temp\f-2024827-35788-15ar1sw.324f; stat'

 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite
  |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf ***@***.***>
  |
 |  Distributed under an ISC-like license, type "license" to see terms
   |
 \----------------------------------------------------------------------------/
 Yosys 0.45+139 (git sha1 8e1e2b9a3, x86_64-w64-mingw32-g++ 13.2.1 -O3)
ERROR: Can't guess frontend for input file `proc;' (missing -f option)!
Perhaps in the configuration of the OSS-CAD-SUITE?

qarlosalberto, 2024-09-27:
Make sure that the focus is in the editor, not un other panel. Put the cursor in the editor and later press the schematic button.

2papyrikoo, 2024-09-27:
I imagine that I put the cursor in the current code right? It keeps saying
yosys failed. i tried it on the command prompt and shows this
[OSS CAD Suite] C:\OSS-CAD-SUITE\oss-cad-suite>
C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p 'read_verilog -sv
"c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\I2C\I2Cslave.v"; hierarchy
-top I2CslaveWith8bitsIO; proc; write_json
C:\Users\jpove\AppData\Local\Temp\f-2024827-35788-1hjiy0u.t8gx; stat'
[00000.000000]
[00000.000381]
 /----------------------------------------------------------------------------\
[00000.000676]  |  yosys -- Yosys Open SYnthesis Suite
                  |
[00000.000970]  |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <
***@***.***>         |
[00000.001215]  |  Distributed under an ISC-like license, type "license" to
see terms        |
[00000.001443]
 \----------------------------------------------------------------------------/
[00000.001660]  Yosys 0.45+139 (git sha1 8e1e2b9a3, x86_64-w64-mingw32-g++
13.2.1 -O3)
[00000.006496] ERROR: Can't guess frontend for input file
`c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\I2C\I2Cslave.v;' (missing -f
option)!

Correct me if I'm mistaking on the given procedure

qarlosalberto, 2024-09-27:
Let me know if this command works fine, and show me the output:

C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p 'read_verilog -sv "c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\I2C\I2Cslave.v";  hierarchy -top I2CslaveWith8bitsIO; proc; write_json C:\Users\jpove\AppData\Local\Temp\f-2024826-5192-87xwap.acpee; stat'

qarlosalberto, 2024-09-27:
@2papyrikoo try with this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

2papyrikoo, 2024-09-27:
I installed and it showed yosys failed. However when I ran the command in
the cmd prompt it worked. Now here is the thing, I set the oss-cad-suite
first and then ran the command. Perhaps running first the command
C:/OSS-CAD-SUITE/oss-cad-suite/start.bat and then the command for yosys, it
could work.

2papyrikoo, 2024-09-28:
Is there a possibility to add an option in the configuration menu, in
schematic viewer where the user adds the path where odd-cad-suite is
installed to set the environment before
(C:\OSS-CAD-SUITE\oss-cad-suite\bin\yosys.exe -p "read_verilog -sv
"c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\I2C\I2Cslave.v"; hierarchy
-top I2CslaveWith8bitsIO; proc; write_json
C:\Users\jpove\AppData\Local\Temp\f-2024828-4036-1h5yye9.hwvb; stat") is
executed?

qarlosalberto, 2024-09-29:
> Is there a possibility to add an option in the configuration menu, in schematic viewer where the user adds the path where odd-cad-suite is installed to set the environment before [...]

In you case you should to configure the "Extra argumentes before..". In `config -> Schematic Viewer -> Extra options passed before...` put: `C:/OSS-CAD-SUITE/oss-cad-suite/start.bat &`

2papyrikoo, 2024-09-29:
I added the command where it said "extra arguments passed before... and
added the & at the end but it seems that it doesnt recognize the path.
Isn't there a way to execute the ./start.bat before executing the yosys.exe
command?

qarlosalberto, 2024-09-29:
Can you show the output? Please use Github code blocks. It's really difficult to read if you use plain text :/

Anyway all the scripts that you add in "extra ..." is exectued before `yosys`. You can check it in the output and adjust the command.

qarlosalberto, 2024-09-29:
Try with:

```
<extracted_location>\oss-cad-suite\environment.bat & 
```

or

```
<extracted_location>\oss-cad-suite\start.bat & 
```

2papyrikoo, 2024-09-30:
It wooooorrkeeeeed 
[image]
I set it as C:\OSS-CAD-SUITE\oss-cad-suite\environment.bat & 
Thanks so much for your support, I really appreciate it

> Can you show the output? Please use Github code blocks. It's really difficult to read if you use plain text :/
Sure, sorry :(

2papyrikoo, 2024-10-01:
Hello again, how can i achieve the schematic viewer for vhdl, the output terminal says: 
`2024-10-01 08:16:43.004 [info] Executing: C:\WINDOWS\system32\cmd.exe  C:\WINDOWS\system32\cmd.exe /d /s /c "C:\OSS-CAD-SUITE\oss-cad-suite\environment.bat & yosys.exe -m ghdl -p "ghdl --std=08 -fsynopsys  --work=work "c:\Users\jpove\OneDrive\Documentos\PROYS\VHDL\main\src\main.vhd"  --work=work -e elevator; hierarchy -top elevator; proc; synth; write_json C:\Users\jpove\.teroshdl_KowTr; stat""
2024-10-01 08:16:43.316 [info] ERROR: Can't load module `./ghdl': "C:\OSS-CA~1\OSS-CA~1\bin\..\share\yosys\plugins\ghdl.so": The specified module could not be found.`

Do i need to add extra argument for ghdl?

qarlosalberto, 2024-10-01:
It should work. Ask in OSS-CAD repository.

qarlosalberto, 2024-10-01:
Also you can use: https://github.com/ghdl/ghdl-yosys-plugin?tab=readme-ov-file#pre-built-packages

qarlosalberto, 2024-10-03:
I close this issue. Feel free to open a new one.

closed 2024-10-03 by qarlosalberto