TerosTechnology/vscode-terosHDL #657: VUnit.project not found, check dependencies check out

WernerFS, 2024-09-10:
**Describe the bug**
VSCode when opening a new project on TerosHDL it will always show in the OUTPUT tab that VUnit.project cannot be found. 

OUTPUT
```
2024-09-10 10:01:58.853 [error] Error while getting dependency graph.
2024-09-10 10:01:58.854 [error] Traceback (most recent call last):
  File "~/.vscode-server/extensions/teros-technology.teroshdl-5.0.12/node_modules/teroshdl2/out/project_manager/dependency/vunit_dependency.py", line 28, in <module>
    import vunit.project as pj
ModuleNotFoundError: No module named 'vunit.project'
```

Check dependencies:
```
2024-09-10 10:05:06.759 [info] Checking dependencies...
2024-09-10 10:05:07.055 [info] ------->  Python found: python3
2024-09-10 10:05:07.056 [info] ------->  Python found in path: /usr/bin/python3
2024-09-10 10:05:07.091 [info] ------->  vunit found 
2024-09-10 10:05:07.324 [info] ------->  cocotb found 
2024-09-10 10:05:07.355 [info] ------->  edalize found 
2024-09-10 10:05:07.364 [info] ------->  Make found in path: make.
```

**Please complete the following information:**
- OS: Ubuntu 22.04.4
- VSCode version version [e.g. 22]
- WSL version: 2.2.4.0
- Kernel version: 5.15.153.1-2
- Python 3.10.12
- cocotb==1.9.1
- edalize==0.5.4
- teroshdl==3.0.0
- vunit==0.1.3
- vunit-hdl==4.7.0

Thanks for taking the time to check this.

rameloni, 2024-09-18:
I have a similar error as well 
```
import vunit.project as pj
ModuleNotFoundError: No module named 'vunit'
```

qarlosalberto, 2024-09-19:
Can you try with this release? https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

WernerFS, 2024-09-21:
Thanks I will give it a go and let you know by next week.

rameloni, 2024-09-23:
> Can you try with this release? https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

@qarlosalberto I tried the release and it works! I don't have that error anymore. Thanks ;)

WernerFS, 2024-09-23:
It's working for me too. Thanks, I will close the issue now.

closed 2024-09-23 by WernerFS