TerosTechnology/vscode-terosHDL #659: Schematic viewer with GHDL/Yosys cannot find work lib

atticlabsdesign, 2024-09-14:
I am using VHDL and when I try and use the schematic viewer it works fine with entities in their own file but when i create an instance of a entity in another file GHDL fails to find the work lib.  This is only a problem for me as it relates to the yosys schematic viewer, I can successfully run test benches with GHDL no problem. I have tried specifying the work directory with an absolute path in the schematic viewer settings (in the arguments passed to GHDL box) and it still fails.

yosys error (without any settings being messed with below)

```
2024-09-13 21:13:32.592 [info] 
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.38+141 (git sha1 078b876f5, clang++ 14.0.0-1ubuntu1.1 -fPIC -Os)

-- Running command `ghdl --std=08 -fsynopsys  --work=work /home/attic_labs/Documents/HDL_PROJ/SCC/Manchester_Encoder_tb.vhd  --work=work -e Manchester_Encoder_tb; hierarchy -top Manchester_Encoder_tb; proc; ; write_json /home/attic_labs/.teroshdl_leKKC; stat' --

1. Executing GHDL.

2024-09-13 21:13:32.603 [info] /home/attic_labs/Documents/HDL_PROJ/SCC/Manchester_Encoder_tb.vhd:23:41: unit "manchester_encoder" not found in library "work"
  Manchester_Encoder_inst : entity work.Manchester_Encoder
                                        ^

2024-09-13 21:13:32.603 [info] ERROR: vhdl import failed.

2024-09-13 21:13:32.605 [error] Yosys failed.

```
See Attached Files for code and settings

 - System: 
Operating System: Kubuntu 22.04
KDE Plasma Version: 5.24.7
KDE Frameworks Version: 5.92.0
Qt Version: 5.15.3
Kernel Version: 6.8.0-40-generic (64-bit)
Graphics Platform: X11
Processors: 12 × Intel® Core™ i7-8750H CPU @ 2.20GHz
Memory: 15.5 GiB of RAM
Graphics Processor: Mesa Intel® UHD Graphics 630

 - VSCode:
Version: 1.93.0
Commit: 4849ca9bdf9666755eb463db297b69e5385090e3
Date: 2024-09-04T13:02:38.431Z
Electron: 30.4.0
ElectronBuildId: 10073054
Chromium: 124.0.6367.243
Node.js: 20.15.1
V8: 12.4.254.20-electron.0
OS: Linux x64 6.8.0-40-generic snap

-TerosHDL:
v6.0.3 (pre-release)

**Screenshots**
If applicable, add screenshots to help explain your problem. **Share the code as text, not as a screenshots!**

[Teros HDL bug report.zip](https://github.com/user-attachments/files/17000413/Teros.HDL.bug.report.zip)

atticlabsdesign, 2024-09-14:
I realize that the config file shows I passed an absolute path to the regular test bench GHDL command, don't worry I realized this was wrong and really did try it in the schematic viewer version and it didnt work, i just forgot to remove that before copying the files

qarlosalberto, 2024-09-21:
Make sure that you aren't using a file with whitespaces in the path. And that your code is synthesizable. For example your testbench will fail because the code is'nt synthesizable.

erik-ski, 2024-12-11:
same here. compilation (I haven't build a proper testbench yet) goes without error, but while wanting to see the schematic, the ghdl invocation does not take into account the existence of my library, it just takes into account my top module

qarlosalberto, 2024-12-11:
@erik-ski are you using Project Schematic or File Schematic? Keep in mind that you need to use Project Schematic if you want to get the schematic for a project.

erik-ski, 2024-12-11:
Sorry. It was another issue.
I just noticed an error while rendering the schematic. I was assigning x"0" to a 5 bit std_logic_vector. Once I changed the assignment for an (others=>'0') it has rendered the circuit.

This takes me to two other questions... 
1. How can a ghdl compilation go through with an error like this?
2. Why doesn't the linter complain?

qarlosalberto, 2024-12-11:
1. It's a question for `ghdl` repo ^^
2. What linter are you using in your backend?

erik-ski, 2024-12-11:
1. Definitely
2. VHDL-LS (I do not know which one is better, nor their differences)

Also, going back to the schemtatic rendering: there are two buttons for rendering in terosHDL: One is on the top right corner, the other one is under "hierarchy", in the left panel. The first one fails, the second one does not fail.

Output for non-failing:
```
2024-12-11 12:30:02.684 [info] Executing: /bin/sh  /bin/sh -c  yosys -m ghdl -p "ghdl --std=08 -fsynopsys -P./helpers --work=helpers "/home/erik/vhdl/acumulador/helpers/helpers_pkg.vhd" --work=work "/home/erik/vhdl/acumulador/acumulador_dp.vhd"  --work=work -e acumulador_dp; hierarchy -top acumulador_dp; proc; write_json /home/erik/.teroshdl_LGuqL; stat"
2024-12-11 12:30:02.693 [info] 
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2024  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.47+116 (git sha1 4b3c03dab, clang++ 18.1.8 -fPIC -O3)

-- Running command `ghdl --std=08 -fsynopsys -P./helpers --work=helpers /home/erik/vhdl/acumulador/helpers/helpers_pkg.vhd --work=work /home/erik/vhdl/acumulador/acumulador_dp.vhd  --work=work -e acumulador_dp; hierarchy -top acumulador_dp; proc; write_json /home/erik/.teroshdl_LGuqL; stat' --

1. Executing GHDL.

2024-12-11 12:30:02.739 [info] /home/erik/vhdl/acumulador/acumulador_dp.vhd:28:10:note: found RAM "weight_memory", width: 5 bits, depth: 10
  signal weight_memory          : weight_memory_t;
         ^

2024-12-11 12:30:02.740 [info] Importing module acumulador_dp.

2024-12-11 12:30:02.740 [info] 
2. Executing HIERARCHY pass (managing design hierarchy).

2.1. Analyzing design hierarchy..
Top module:  \acumulador_dp

2024-12-11 12:30:02.740 [info] 
2.2. Analyzing design hierarchy..
Top module:  \acumulador_dp
Removed 0 unused modules.

3. Executing PROC pass (convert processes to netlists).

3.1. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

3.2. Executing PROC_RMDEAD pass (remove dead branches from decision trees).
Removed a total of 0 dead cases.

2024-12-11 12:30:02.740 [info] 
3.3. Executing PROC_PRUNE pass (remove redundant assignments in processes).
Removed 0 redundant assignments.
Promoted 0 assignments to connections.

3.4. Executing PROC_INIT pass (extract init attributes).

3.5. Executing PROC_ARST pass (detect async resets in processes).

3.6. Executing PROC_ROM pass (convert switches to ROMs).
Converted 0 switches.

3.7. Executing PROC_MUX pass (convert decision trees to multiplexers).

3.8. Executing PROC_DLATCH pass (convert process syncs to latches).

2024-12-11 12:30:02.740 [info] 
3.9. Executing PROC_DFF pass (convert process syncs to FFs).

2024-12-11 12:30:02.740 [info] 
3.10. Executing PROC_MEMWR pass (convert process memory writes to cells).

3.11. Executing PROC_CLEAN pass (remove empty switches from decision trees).
Cleaned up 0 empty switches.

[... 47 lines cut ...]
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.47+116 (git sha1 4b3c03dab, clang++ 18.1.8 -fPIC -O3)

-- Running command `ghdl --std=08 -fsynopsys  --work=work /home/erik/vhdl/acumulador/acumulador_dp.vhd  --work=work -e acumulador_dp; hierarchy -top acumulador_dp; proc; write_json /home/erik/.teroshdl_lWdsy; stat' --

1. Executing GHDL.

2024-12-11 12:33:50.150 [info] /home/erik/vhdl/acumulador/acumulador_dp.vhd:5:9:error: cannot find resource library "helpers"
library helpers;
        ^
/home/erik/vhdl/acumulador/acumulador_dp.vhd:6:13:error: unit "helpers_pkg" not found in library "helpers"
use helpers.helpers_pkg.all;
            ^

2024-12-11 12:33:50.150 [info] /home/erik/vhdl/acumulador/acumulador_dp.vhd:13:56:error: no declaration for "clog2"
    AER                          : in std_logic_vector(clog2(nweights) - 1 downto 0);
                                                       ^
/home/erik/vhdl/acumulador/acumulador_dp.vhd:19:73:error: no declaration for "clog2"
  constant nbits_addr    : integer                                   := clog2(nweights);
                                                                        ^
ERROR: vhdl import failed.

2024-12-11 12:33:50.151 [error] Yosys failed.
```

qarlosalberto, 2024-12-11:
The best is `VHDL-LS`. Then for your second question you have to ask in: https://github.com/VHDL-LS/rust_hdl

The button on the editor only render the current open file. I doesn't "know" about other files in the project.
The buttion in under "hierarchy" take all the project files and it renders the schematic.

erik-ski, 2024-12-11:
Thank you very much