TerosTechnology/vscode-terosHDL #660: Q: TerosHDL and Python virtual environments

mkaiser, 2024-09-17:
Hi, 

I have a rather special use case for running TerosHDL and having trouble setting up the toolchain:

[image]

As you can see, I am using the SSH feature to connect to a remote server for development and can't simply start VSCode on my Notebook with a customized virtual environment (therefore the use-cases I found on the docs / github do not match :/ ). 

**Is there any way to "tell" TerosHDL to use a specific Python venv?**

I tried this, having success in finding vunit, cocolib and edalize, but for example, I cannot properly set the installation path for GHDL: 

[image]

[image]

Additional info: 

content of "setup_teros_venv.sh" >>

```
TEROS_DIR="hw/TerosHDL"
PY_CMD=$(which python3)
TEROS_VENV_DIR="${TEROS_DIR}/venv"

$PY_CMD -m venv $TEROS_VENV_DIR && source $TEROS_VENV_DIR/bin/activate && pip install -r $TEROS_DIR/requirements.txt

echo -e "\nConfigure Teros: Click on the Teros icon on the left --> General --> Open Configuration Menu"
echo "For Python3 binary path add the path to the venv folder, e.g.," 
echo -e "$TEROS_VENV_DIR/bin/python3" 

```
content of "requirements.txt" >>
```
vunit
edalize 
cocotb 
teroshdl
vsg
yowasp-yosys
ghdl

```

qarlosalberto, 2024-09-17:
Maybe I don't understand it good. Are you installing GHDL with pip?

mkaiser, 2024-09-17:
Yes, I did install IT with pip. Was this bad idea? 

I have no sudo permissions on the synthesis machine, so installing there is tedious administrative stuff. I hope this could be done somehow via pyvenv

qarlosalberto, 2024-09-17:
Try using the directory path instead of the binary path for GHDL. And
double check that the directory contains the binary.

jevogel, 2025-02-28:
@mkaiser, I had a similar experience with TerosHDL and Python virtual environments. See my comment here: #729

JennySmith888, 2025-03-27:
I think I'm also having issues with this? I am running VS code locally but my VS Code workspace is configured to use a python virtual environment and that same venv is specified in the Teros HDl general settings Python3 binary path. When I run "Verify Setup" 

vsg is found initially
```
⊙ Checking Python dependencies. Current configured installation path: "/home/jpsmith/python_venvs/generic_fpga/bin/python3"
  🎉 vunit found.
  🎉 edalize found.
  🎉 cocotb (optional) found.
  🎉 vsg (optional) found.
```
But the same check fails during the Formatter configuration check?
```⊙ Checking formatter vsg for VHDL.
  🔎 Searching for the binary "Vsg" in the system path
  ❌ Vsg not found in the system path. Search executed with: "vsg --version"
👎 The linter installation path is not correctly configured. The binary could not be located in either the system path or the configured path. Please check the documentation.```

I'm not clear if vsg linting is working as expected or if my custom configuration is being used.

qarlosalberto, 2025-03-28:
@JennySmith888 ahh okok got it. 

1. Do an import in Python: `import vsg`. Your Python is correctly configured so it will work.
2. Do "vsg --version" in the command line. But `vsg` binary is in you venv. So it will fail.

I think that the first check doesn't make sense. So I will delete it.

You can configure the VSG binary path in:

[image]

** Yes there are 3 tab to configure VSG :/ I will fix it soon, I have to think in the best way to do it.

qarlosalberto, 2025-03-28:
Try with this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest