TerosTechnology/vscode-terosHDL #661: Downgrades from v6.0.1 to v6.0.3

ArnePret, 2024-09-19:
**Describe the bug**
The following two things worked flawlessly for me in v6.0.1 and have buggy behavior in v6.0.3.:
1. use work.SOMETHING.all; is only found by "vhdl ls" if the corresponding file was opened manually. (TerosProject is not searched, neither the current folder structure)
2. library SELFCOMPILED;
        context SLEFCOMPILED.whatever;
This library is not found at all. 

**To Reproduce**
1. Create the package/component for your entity/architecture in a seperate file and use it. Add both files to a Teros Project:
///// File for package "MODULE_pkg.vhd":
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;
  use ieee.math_real.all;

package MODULE_pkg is

  constant CONST : natural := 1;

  component MODULE is
    port (
      INPUT  : in    std_logic;
      OUTPUT : out   std_logic
    );
  end component MODULE;

end package MODULE_pkg;

///// File for entity "MODULE.vhd":
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;
  use ieee.math_real.all;
  use work.MODULE_pkg.all;

entity MODULE is
  port (
    INPUT  : in    std_logic;
    OUTPUT : out   std_logic
  );
end entity MODULE;

architecture arch of MODULE is

begin

  OUTPUT <= INPUT;

end architecture arch;

2. Compile osvvm and include the following:
library osvvm;
  context osvvm.OsvvmContext;

**Please complete the following information:**
 - OS: Windows 11
 - VSCode version version 1.92.2

qarlosalberto, 2024-09-19:
I don't understand your problem. What is the error? OSVVM is not included? You need to add it to the project.

[image]

ArnePret, 2024-09-19:
I am sorry, I have phrased this badly.
First of all: These are two different problems (I should have probably created two issues for that). I put them into one Issue, because they seem related to me.
Second: The code from your screenshot looks like this for me (on v6.0.3) but only if i have not opened MODULE_pkg yet. (Every time I open VSCode) (This was automatically detected in v6.0.1)
Third: I have also added the osvvm library (the .vhd files, because the compiled .lib didn't work for me in either version).

The Module_pkg error is solved by opening the MODULE_pkg.vhd (but this has to be done every time VSCode is opend). The osvvm error I was unable to solve in v6.0.3. This worked both without issue in v6.0.1.

[image]

qarlosalberto, 2024-09-19:
Can you try with this release? https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

ArnePret, 2024-09-19:
Everything works with this version. VSCode reports both as v6.0.3, but the one installed via VSCode produces the errors.
Thanks for the quick help.

qarlosalberto, 2024-09-23:
Can you show me a screenshot with your project and files view? Similar to my previous screenshot

ArnePret, 2024-09-23:
I updated to Teros v6.0.4 and restarted VSCode.
I updated the code, closed everything and restarted VSCode.
And everything works now. I have no Idea why it didn't work before (maybe VSCode needs 3 restarts to load everything properly?) I am sorry to have wasted your time with this again. Also in my real project everything is detected correctly.
Keep up the great work

closed 2024-09-23 by ArnePret