TerosTechnology/vscode-terosHDL #668: Spaces in Project Name Broken?

jchang-endiag, 2024-09-26:
**Describe the bug**
It appears that after TerosHDL automatically updated, spaces in project names no longer work.  My projects all have a spaces, used to separate the local repo copy and the project name and they no longer show up in the PROJECTS window.

**To Reproduce**
1. Click _Projects > +_ (Add project)
2. Click _Create and empty Generic project_
3. Type a name with spaces, such as "test project"
4. Observe "test project" does not appear in PROJECTS window.

**Please complete the following information:**
 - OS: Microsoft Windows 10 Pro, 10.0.19045 Build 19045
 - VSCode version: 1.93.1

**Screenshots**
n/a

**Additional context**
none

closed 2024-09-26 by qarlosalberto

jchang-endiag, 2024-09-30:
Thanks!  Unfortunately, my old projects are still MIA.  Perhaps this is a side effect of switching to per project configuration?

qarlosalberto, 2024-10-01:
> Thanks! Unfortunately, my old projects are still MIA. Perhaps this is a side effect of switching to per project configuration?

In theory it should work... :/

PR #669 (2024-09-26): feat: allow white spaces in project name
fixes #668