TerosTechnology/vscode-terosHDL #677: Failed to access library 'C:\Users\namhe\.teroshdl_fryIl_' at "C:\Users\namhe\.teroshdl_fryIl_".TerosHDL: modelsim(vlog-19)

FinnNGrace, 2024-10-01:
It's look like there is some errors.
[image]

Here is the folder:
[image]

TopplyZ, 2025-01-09:
Did you solve it? I had the same problem.

PeterMonsson, 2025-02-04:
1) Open the TerosHDL extension in the sidebar
2) Select Open Global Settings Menu in the bottom-ish of the sidebar
3) Go to Linter Settings
4) Set Verilog/SV Linter to Disabled

PedroFPGA, 2025-03-13:
This is happening to me as well. It was working fine not so long ago but had to reinstall OS recently and now I’m getting this error all the time

qarlosalberto, 2025-03-20:
1. Install this version: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest 
2. Open a file, and save it.
3. Go to OUTPUT->TerosHDL:Debug and copy paste the text. Also share a screenshot. It will be similar to:

[image]