TerosTechnology/vscode-terosHDL #679: Quartus crashes when compiling

ecstrema, 2024-10-03:
I just imported a quartus project into TerosHDL, and I can't manage to compile it with TerosHDL, although it works from Quartus.

I get the following error:

```
Info: Elaborating from top-level entity "VfcHdTop"
Internal Error: Sub-system: QIS, File: /quartus/synth/qis/qis_rtl_stage_elaborator.cpp, Line: 531
!dms_design.expired()
Stack Trace:
  Quartus          0x46f73: QIS_RTL_STAGE::IMPL::elaborate + 0x1953 (synth_qis)
  Quartus          0x1ddc5: qis_elaborate + 0x1b5 (synth_qis)
  Quartus          0x16442: TclNRRunCallbacks + 0x62 (tcl86)
  Quartus          0x17c4d: TclEvalEx + 0x9ed (tcl86)
  Quartus          0xa6a8b: Tcl_FSEvalFileEx + 0x22b (tcl86)
  Quartus          0xa5136: Tcl_EvalFile + 0x36 (tcl86)
  Quartus          0x151f6: qexe_evaluate_tcl_script + 0x4e6 (comp_qexe)
  Quartus          0x140eb: qexe_do_tcl + 0x46b (comp_qexe)
  Quartus          0x1a67e: qexe_run_tcl_option + 0x5ee (comp_qexe)
  Quartus          0x14b81: QCU::DETAIL::intialise_qhd_and_run_qexe + 0xa1 (comp_qcu)
  Quartus          0x214a2: qcu_run_tcl_option + 0x2f2 (comp_qcu)
  Quartus           0x12cc: qsyn2_tcl_process_default_flow_option + 0x1dc (quartus_syn)
  Quartus          0x19f6b: qexe_run + 0x3eb (comp_qexe)
  Quartus          0x1b0ba: qexe_standard_main + 0x26a (comp_qexe)
  Quartus           0x2fe1: qsyn2_main + 0x131 (quartus_syn)
  Quartus          0x190b8: msg_main_thread + 0x18 (CCL_MSG)
  Quartus          0x19781: msg_thread_wrapper + 0x71 (CCL_MSG)
  Quartus          0x22dc0: mem_thread_wrapper + 0x70 (ccl_mem)
  Quartus          0x1734d: msg_exe_main + 0x20d (CCL_MSG)
  Quartus           0x4b68: __scrt_common_main_seh + 0x11c (quartus_syn)
  Quartus          0x17373: BaseThreadInitThunk + 0x13 (KERNEL32)
  Quartus          0x4cc90: RtlUserThreadStart + 0x20 (ntdll)

End-trace
```

# Reproduction

I have no small reproduction, and my repo is in CERN hands, so it will probably be open one day, but is not right now.

**Please complete the following information:**
 - OS: Windows 10
 - VSCode version: 1.92.2

Important note: I have quartus 21.2, which is an older version. However, compiling works perfectly with Quartus, but not from TerosHDL...

<details>

```
2024-10-03 15:16:15.967 [info] Critical Warning (19941): QSF command 'set_parameter' can only apply to the top-level design and cannot include '-to' or '-entity' arguments. File: c:/Users/rmarche/cernbox/Documents/repos/eavp-std/hdl/synthesis/quartus_prj/VfcHd_Application.qsf Line: 83
Critical Warning (19941): QSF command 'set_parameter' can only apply to the top-level design and cannot include '-to' or '-entity' arguments. File: c:/Users/rmarche/cernbox/Documents/repos/eavp-std/hdl/synthesis/quartus_prj/VfcHd_Application.qsf Line: 84
Critical Warning (19941): QSF command 'set_parameter' can only apply to the top-level design and cannot include '-to' or '-entity' arguments. File: c:/Users/rmarche/cernbox/Documents/repos/eavp-std/hdl/synthesis/quartus_prj/VfcHd_Application.qsf Line: 85
Critical Warning (19941): QSF command 'set_parameter' can only apply to the top-level design and cannot include '-to' or '-entity' arguments. File: c:/Users/rmarche/cernbox/Documents/repos/eavp-std/hdl/synthesis/quartus_prj/VfcHd_Application.qsf Line: 86
Info: *******************************************************************
Info: Running Quartus Prime Synthesis
    Info: Version 21.2.0 Build 72 06/14/2021 SC Pro Edition
    Info: Copyright (C) 2021  Intel Corporation. All rights reserved.
    Info: Your use of Intel Corporation's design tools, logic functions 
    Info: and other software and tools, and any partner logic 
    Info: functions, and any output files from any of the foregoing 
    Info: (including device programming or simulation files), and any 
    Info: associated documentation or information are expressly subject 
    Info: to the terms and conditions of the Intel Program License 
    Info: Subscription Agreement, the Intel Quartus Prime License Agreement,
    Info: the Intel FPGA IP License Agreement, or other applicable license
    Info: agreement, including, without limitation, that your use is for
    Info: the sole purpose of programming logic devices manufactured by
    Info: Intel and sold by Intel or its authorized distributors.  Please
    Info: refer to the applicable agreement for further details, at
    Info: https://fpgasoftware.intel.com/eula.
    Info: Processing started: Thu Oct  3 15:15:18 2024
[... 88 lines cut ...]
  Quartus          0x16442: TclNRRunCallbacks + 0x62 (tcl86)
  Quartus          0x17c4d: TclEvalEx + 0x9ed (tcl86)
  Quartus          0xa6a8b: Tcl_FSEvalFileEx + 0x22b (tcl86)
  Quartus          0xa5136: Tcl_EvalFile + 0x36 (tcl86)
  Quartus          0x151f6: qexe_evaluate_tcl_script + 0x4e6 (comp_qexe)
  Quartus          0x140eb: qexe_do_tcl + 0x46b (comp_qexe)
  Quartus          0x1a67e: qexe_run_tcl_option + 0x5ee (comp_qexe)
  Quartus          0x14b81: QCU::DETAIL::intialise_qhd_and_run_qexe + 0xa1 (comp_qcu)
  Quartus          0x214a2: qcu_run_tcl_option + 0x2f2 (comp_qcu)
  Quartus           0x12cc: qsyn2_tcl_process_default_flow_option + 0x1dc (quartus_syn)
  Quartus          0x19f6b: qexe_run + 0x3eb (comp_qexe)
  Quartus          0x1b0ba: qexe_standard_main + 0x26a (comp_qexe)
  Quartus           0x2fe1: qsyn2_main + 0x131 (quartus_syn)
  Quartus          0x190b8: msg_main_thread + 0x18 (CCL_MSG)
  Quartus          0x19781: msg_thread_wrapper + 0x71 (CCL_MSG)
  Quartus          0x22dc0: mem_thread_wrapper + 0x70 (ccl_mem)
  Quartus          0x1734d: msg_exe_main + 0x20d (CCL_MSG)
  Quartus           0x4b68: __scrt_common_main_seh + 0x11c (quartus_syn)
  Quartus          0x17373: BaseThreadInitThunk + 0x13 (KERNEL32)
  Quartus          0x4cc90: RtlUserThreadStart + 0x20 (ntdll)

End-trace
```

</details>

qarlosalberto, 2024-10-03:
Sadly, I cannot guarantee that it works well for versions lower than `Quartus Pro 23.3`

qarlosalberto, 2024-10-03:
Try to clean the project.

ecstrema, 2024-10-03:
Of course, I understand. I'll close this as not planned. 

I've tried cleaning the project, the same thing happens.

closed 2024-10-03 by ecstrema