TerosTechnology/vscode-terosHDL #685: Recommended flow for standalone CLI documentor

davidgussler, 2024-10-15:
I came across the [teroshdl-documenter-demo repo](https://github.com/TerosTechnology/teroshdl-documenter-demo), which has some examples for generating documentation on the CLI for a CI pipeline. I would like to set up something like this, but I noticed that the standalone [cloibri project](https://github.com/TerosTechnology/colibri), which is used by the documenter demo, has been archived and that there is a [colibri directory](https://github.com/TerosTechnology/vscode-terosHDL/tree/dev/src/colibri) in the main [vscode-terosHDL repo](https://github.com/TerosTechnology/vscode-terosHDL). The documentor demo repo has also not been touched in 3 years.

Has the colibri project been merged with the main terosHDL project? Is there new recommended way to use the documentor on the CLI?

Thanks for your support! The documenter is a remarkable tool and I would love to be able to automate it without needing to open up VSCode to click a button.

monkey265, 2024-11-12:
I think that right now CLI is not supported.

```
$ ./bin/dev teroshdl
(node:3361844) Error Plugin: teroshdl: files attribute must be specified in /home/jcada/Downloads/teros/vscode-terosHDL/package.json
module: @oclif/core@1.26.2
plugin: teroshdl
root: /home/jcada/Downloads/teros/vscode-terosHDL
See more details with DEBUG=*
(Use `node --trace-warnings ...` to show where the warning was created)
Error: command teroshdl not found
    at Config.runCommand (/home/jcada/Downloads/teros/vscode-terosHDL/node_modules/@oclif/core/lib/config/config.js:271:19)

```

In the colibri repo, just formatter, linter and template are usable.
```
 ./bin/dev teroshdl
Check errors in HDL files.

USAGE
  $ teroshdl teroshdl:COMMAND

COMMANDS
  teroshdl:formatter  Check errors in HDL files.
  teroshdl:linter     Check errors in HDL files.
  teroshdl:template   Generate HDL template from a file.

```

So unfortunately, it seems like documentation generation via CLI is not supported right now at all.

monkey265, 2024-11-12:
As an alternative you can use following tools:

https://kevinpt.github.io/symbolator/
https://github.com/kevinpt/hdlparse

eric-hologic, 2025-03-07:
We may have some interest in contributing to the project to re-enable command line documentation generation.  Would one of the existing devs be willing to give a quick overview of the commandline system and what needs to be fixed to re-enable this?

qarlosalberto, 2025-03-07:
@eric-hologic send me an email and we can discuss it: carlosruiznaranjo@gmail.com

qarlosalberto, 2025-09-22:
No. I closed it accidentally.