TerosTechnology/vscode-terosHDL #687: VHDL modules using Verilog libraries - library not found

rameloni, 2024-10-23:
Hello, I have a VHDL module that imports and uses a Verilog library. Even though I setup the project properly, Teros is not able to recognize the verilog libraries and modules.

How can I do that?

qarlosalberto, 2024-10-23:
Sadly it's not possible. TerosHDL recognizes VHDL and Verilog/SV independently.

rameloni, 2024-10-23:
Alright, thank you! Is there an idea of supporting it in Teros?

qarlosalberto, 2024-10-23:
Not in the near future, it's s big task.

rameloni, 2024-10-23:
I see I understand. Apart from that, I'm also facing an issue with vendors IPs. For example, when I try to import a Xilinx library teros reports "not found". 
I'm mainly using teros for its linting and checking features. Maybe, it would be nice to mark something as "blackbox" or simply ask teros to ignore it. What do you think?

rameloni, 2024-10-23:
Yes https://github.com/TerosTechnology/vscode-terosHDL/issues/688