TerosTechnology/vscode-terosHDL #691: Python Virtual Environments not Usable

thomas-woehrle, 2024-11-01:
**Describe the bug**
When one uses a python virtual environment to install the necessary dependencies, the schematic viewer does not work. 

**Code**
Creating pyenv virtualenv (assuming pyenv virtualenv plugin is installed):
```sh
pyenv virtualenv 3.x.x teroshdl
pyenv activate teroshdl
pip install vunit-hdl edalize yowasp-yosys cocotb
```

Then referencing the corresponding binary in the extension settings, fe:
```/Users/<username>/.pyenv/versions/3.12.7/envs/teroshdl/bin/python```

Now, the dependency check goes through, but the schematic viewer throws this: 
```
2024-11-01 17:41:01.885 [info] Executing: /bin/sh  /bin/sh -c  yowasp-yosys -p "read_verilog -sv "/<path>/add.v"; hierarchy -top add; proc; write_json .teroshdl_eOEbL; stat"
2024-11-01 17:41:01.987 [info] pyenv: yowasp-yosys: command not found

2024-11-01 17:41:02.065 [info] 
The `yowasp-yosys' command exists in these Python versions:

2024-11-01 17:41:02.066 [info]   3.12.7/envs/teroshdl
  teroshdl

2024-11-01 17:41:02.066 [info] 
Note: See 'pyenv help global' for tips on allowing both

2024-11-01 17:41:02.067 [info]       python2 and python3 to be found.

2024-11-01 17:41:02.067 [error] Yosys failed.
```

**Please complete the following information:**
 - OS: MacOS 14.2.1
 - VSCode version 1.95.1

**Additional context**
I tried installing the dependencies in a global python binary and referencing that instead and it worked, so I assume that nothing else in my setup is broken.

qarlosalberto, 2024-11-01:
In the schematic settings use the pre command option to activate your Python environment.

thomas-woehrle, 2024-11-01:
Thank you @qarlosalberto that worked, however I had to do ```source ~/.zshrc && pyenv activate teroshdl &&```, without the ```&&``` at the end it didn't work, not sure if it is intended that the specified command is just inserted

thomas-woehrle, 2024-11-01:
Also another question: Looking back my bug was not really a bug but a question and I have another question regarding syntax checking which is not a bug but just a question as well. 

Is there a better place to ask these things?

closed 2024-11-01 by qarlosalberto