TerosTechnology/vscode-terosHDL #693: Cannot run test benches for Verilog / render schematic views

washburn961, 2024-11-02:
I coded up a simple Verilog module and added it to a project on TerosHDL:

[image]

When I hover my mouse over the first line I see:

[image]

Maybe this is related to ModelSim since I have it installed? No idea.
When I hit the "Play" button in order to run my testbench, this is the log I obtain:

`
2024-11-01 22:11:18.231 [info] c:\Users\Gabriel\Desktop\FPGA\test.v has unknown file type 'verilogSource'
c:\Users\Gabriel\Desktop\FPGA\test_tb.v has unknown file type 'verilogSource'

2024-11-01 22:11:18.293 [info] Creating libraries directories

2024-11-01 22:11:18.340 [info] Creating libraries directories

2024-11-01 22:11:18.342 [info] ghdl -m --std=93c   and_gate_tb

2024-11-01 22:11:18.354 [info] C:\msys64\mingw64\bin\ghdl.exe:error: cannot find entity or configuration and_gate_tb

2024-11-01 22:11:18.355 [info] make: *** [Makefile:16: and_gate_tb] Error 1

2024-11-01 22:11:18.357 [info] 
************************************************************************************************
---> Build directory: C:\Users\Gabriel\.teroshdl\build
---> Make installation folder path: System path
Error: '['make', 'run']' exited with an error: 2
`

I am not sure why it's trying to use GHDL. Is it possible to run the simulation on ModelSim and view the waveforms on TerosHDL?
I have done the following:

- Installed TerosHDL;
- Had already Python 3 installed and added to PATH;
- Installed the Python 3 dependencies;
- Installed "make" via Chocolatey, as it is shown in the docs. It works as expected when running `make --version` from windows CMD or Shell;
- Installed oss-cad-suite and added `C:\oss-cad-suite\bin` to PATH;
- Added `C:/oss-cad-suite/start.bat` as an extra option on Global Settings > Schematic viewer, as shown:

[image]

No further configurations were performed.

I have no idea how to proceed with this issue and would appreciate if some assitance could be provided so I can use this tool, because it seems very good.

washburn961, 2024-11-02:
Okay, I just noticed I needed to change from GHDL to ModelSim here:

[image]

Now I have this log output:

```
2024-11-01 22:25:35.951 [info] C:/intelFPGA/20.1/modelsim_ase/win32aloem/vsim -c -do "do edalize_main.tcl; exit"

2024-11-01 22:25:37.221 [info] Reading pref.tcl

2024-11-01 22:25:37.244 [info] 
# 2020.1

2024-11-01 22:25:37.298 [info] # do edalize_main.tcl

2024-11-01 22:25:37.644 [info] #  exit

2024-11-01 22:25:39.615 [info] C:/intelFPGA/20.1/modelsim_ase/win32aloem/vsim -c     -do "run -all; quit -code [expr [coverage attribute -name TESTSTATUS -concise] >= 2 ? [coverage attribute -name TESTSTATUS -concise] : 0]; exit" and_gate_tb

2024-11-01 22:25:40.260 [info] Reading pref.tcl

2024-11-01 22:25:40.274 [info] 
# 2020.1

2024-11-01 22:25:41.255 [info] # vsim -c -do "run -all; quit -code [expr [coverage attribute -name TESTSTATUS -concise] >= 2 ? [coverage attribute -name TESTSTATUS -concise] : 0]; exit" and_gate_tb 
# Start time: 22:25:40 on Nov 01,2024
# Loading work.and_gate_tb
# Loading work.and_gate

2024-11-01 22:25:41.258 [info] # run -all

2024-11-01 22:25:41.277 [info] # A B | Y
# --------
# 0 0 | 0
# 0 1 | 0
# 1 0 | 0
# 1 1 | 1
# ** Note: $finish    : c:/Users/Gabriel/Desktop/FPGA/test_tb.v(35)
#    Time: 40 ps  Iteration: 0  Instance: /and_gate_tb

2024-11-01 22:25:41.281 [info] # End time: 22:25:41 on Nov 01,2024, Elapsed time: 0:00:01
# Errors: 0, Warnings: 0

2024-11-01 22:25:41.357 [info] 
************************************************************************************************
---> Build directory: C:\Users\Gabriel\.teroshdl\build
---> Make installation folder path: 
---> Modelsim directory: C:/intelFPGA/20.1/modelsim_ase/win32aloem
```

But I see no waveforms here:

[image]

qarlosalberto, 2025-03-18:
You can configure Tool GUI or waveform:

[image]

closed 2025-03-18 by qarlosalberto