TerosTechnology/vscode-terosHDL #699: [Question] How to use testbenches?

thomas-woehrle, 2024-11-08:
Hi, 

I want to use Teros' Testbench capability. The documentation thinly shows [how to create testbench templates](https://terostechnology.github.io/terosHDLdoc/docs/guides/templates), but I didn't find more in the documentation about Testbenches and neither in the GitHub issues - feel free to refer me if there is existing documentation. 

I am interested how I can use the Testbenches feature and what it is capable of doing. Can I aggregate different testbenches here and execute them centrally?

Screenshot from VScode currently:
[image]

Thanks in advance and as for the last issue I created (#698) : If I end up knowing well how the feature works after I have worked with it for a while, I will gladly contribute to the documentation.

thomas-woehrle, 2024-11-08:
> […](#)

I have worked through all those documentation pages before, they are not of great help to me unfortunately :/. Mainly because of the issue that I don't even know what sources and libraries can be used for (see this issue #698).

thomas-woehrle, 2024-11-08:
> You can follow this tutorial: https://terostechnology.github.io/terosHDLdoc/docs/tutorials/tutorial_vhdl 

Sorry, but unfortunately this doesn't help either, because it not thorough enough imo. 

This is its explanation of running a simulation: 
[image]

And for me that leaves a lot of things open, fe how to even get a test which can be run in the first place...

I greatly appreciate your work with the extension, but I find it tough to use

qarlosalberto, 2024-11-08:
But I don't understand your question. Do you want to run a simulation? Or are you learning HDL?

thomas-woehrle, 2024-11-08:
I want to run a simulation and I am wondering how TerosHDL is intended to help with that. 

Lets say I have a testfile TestBench.v and I am manually running it like so: 
```bash
iverilog -o test_executable TestBench.v
vvp test_executable
``` 

Is Teros gonna help automize that or can it instead be used f.e. for triggering cocotb tests or ... ?

I don't know what I can use it for and how I can set up tests with it.

qarlosalberto, 2024-11-08:
Then you have to follow this tutorial: https://terostechnology.github.io/terosHDLdoc/docs/tutorials/tutorial_vhdl

You only need to use Icarus instead GHDL.
Let me know if you are stucks in any step.

thomas-woehrle, 2024-11-08:
Okay, I followed the tutorial. 

I set my TestBench file (FloatAdditionTB.v) as top module file and ran the TestBench from the Teros UI. 

I get this error:
```bash
2024-11-08 21:21:25.305 [info] <some_path>/sim/FloatAdditionTB.v has unknown file type 'verilogSource'
 has unknown file type 'none'

2024-11-08 21:21:25.318 [info] Creating libraries directories

2024-11-08 21:21:25.327 [info] Creating libraries directories

2024-11-08 21:21:25.327 [info] ghdl -m --std=93c   FloatAdditionTB

2024-11-08 21:21:25.386 [info] pyenv: ghdl: command not found

2024-11-08 21:21:25.463 [info] 
The `ghdl' command exists in these Python versions:

2024-11-08 21:21:25.464 [info]   3.12.7/envs/teroshdl
  teroshdl

Note: See 'pyenv help global' for tips on allowing both
      python2 and python3 to be found.

2024-11-08 21:21:25.465 [info] make: *** [FloatAdditionTB] Error 127

2024-11-08 21:21:25.465 [info] 
************************************************************************************************
---> Build directory: <some_path>/.teroshdl/build
---> Make installation folder path: System path
Error: '['make', 'run']' exited with an error: 2
```

> You only need to use Icarus instead GHDL. Let me know if you are stucks in any step.

I have Icarus set up as a tool in Teros

qarlosalberto, 2024-11-08:
It seems that you have GHDL. Your linter is Icarus, but your simulation tool is GHDL.

qarlosalberto, 2024-11-08:
Go to config, tools and select Icarus

thomas-woehrle, 2024-11-08:
Sorry, I am not sure what you mean. 

Do you mean this:
[image]

I can't select it as simulation tool here

qarlosalberto, 2024-11-08:
Click in "Tools". There is another tab.

thomas-woehrle, 2024-11-08:
> Click in "Tools". There is another tab. El vie, 8 nov 2024, 21:39, thomas-woehrle ***@***.***> escribió:

Okay, now it works.

Additional question though:

What do these settings do: 
[image]

Switching between execution mode "Command Line" and "GUI" didn't change anything. 
And the TestBench should create a .vcd (`$dumpfile("FloatAdditionTB.vcd");`), but gtkwave is not run atm. Not sure if this would be intended