TerosTechnology/vscode-terosHDL #701: Vivado version not detected correctly when verifying the setup

fuzesa, 2024-11-11:
**Describe the bug**
When the user clicks the `Verify Setup` button within the _Configuration_ section, it attempts to get the version of the installed Vivado application that was provided in the settings. The problem is that in https://github.com/TerosTechnology/vscode-terosHDL/blob/e2c4806782f8d3290665f926d173aa375eec6ee1/src/teroshdl/features/configChecker/externalTools.ts#L42 file, the `--version` flag is hardcoded, but the two dashes don't work, because the Vivado executable requires only a single dash in order to print the correct version (please see screenshot below). This leads to `Verify Setup` failing. (See second screenshot)

**To Reproduce**
Set the correct path to the Vivado executable and afterwards click on the `Verify Setup` button.

**Please complete the following information:**
 - OS: Arch Linux
 - VSCode version 1.95.2

**Screenshots**
[image]
[image]

fuzesa, 2024-11-11:
Perhaps some modifications similar to this could be added:

https://github.com/fuzesa/vscode-terosHDL/commit/80044f3912fbed14185d90203aee222c9fef81c4

el3ctrician, 2024-12-11:
i confrim the issue, maybe you can open a pull request ?

qarlosalberto, 2025-01-22:
Solved, the new version will be released to the market soon.

closed 2025-01-22 by qarlosalberto