TerosTechnology/vscode-terosHDL #706: ModelSim external tool incorrect version check when verifying setup

voltavidTony, 2024-12-01:
**Describe the bug**
TerosHDL uses wrong binary name and argument to find ModelSim version. `modelsim` does not exist as a binary, and `--version` is wrong.

**To Reproduce**
Obtain ModelSim-Intel from the Intel FPGA Software Download Center.
Configure ModelSim as external tool and set correct install path.
Apply settings.

**Please complete the following information:**
 - OS: Ubuntu 22.04.5
 - VSCode version: 1.95.3
 - TerosHDL version: 6.0.14

**Screenshots**
```
*************************************************
Checking External Tool Configuration (https://terostechnology.github.io/terosHDLdoc/docs/external_tools/) 
❌ Some checks failed ❌
*************************************************
⊙ Selected external tool: MODELSIM. Installation path: "/home/volta/intelFPGA/20.1/modelsim_ase/linux"
  🔎 Searching for the binary modelsim at "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim"
  ❌ modelsim not found at "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim". Search executed with: "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim --version"
  🔎 Searching for the binary modelsim at "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim.exe"
  ❌ modelsim not found at "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim.exe". Search executed with: "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim.exe --version"
  🔎 Searching for the binary modelsim at "/home/volta/intelFPGA/20.1/modelsim_ase/linux"
  ❌ modelsim not found at "/home/volta/intelFPGA/20.1/modelsim_ase/linux". Search executed with: "/home/volta/intelFPGA/20.1/modelsim_ase/linux --version"
  🔎 Searching for the binary modelsim at "/home/volta/intelFPGA/20.1/modelsim_ase/linux.exe"
  ❌ modelsim not found at "/home/volta/intelFPGA/20.1/modelsim_ase/linux.exe". Search executed with: "/home/volta/intelFPGA/20.1/modelsim_ase/linux.exe --version"
  🔎 Searching for the binary "modelsim" in the system path
  ❌ modelsim not found in the system path. Search executed with: "modelsim --version"
👎 The MODELSIM installation path is not correctly configured. The binary could not be located in either the system path or the configured path. Please check the documentation.
```

**Additional context**
There are actually two issues here:
1. ModelSim-Intel does not ship with a 'modelsim' binary. It is called 'vsim' instead
2. The correct command line option is `-version`, not `--version`. TerosHDL adds an extra dash

**Solution**
The solution was to place the following script (a file named 'modelsim' with no extension) in the binary folder:
```
#!/bin/sh
dir=`dirname "$0"`
if [ "$*" = "--version" ] ; then
  exec "$dir/vsim" "-version"
else
  exec "$dir/vsim" "$@"
fi
```
This fixes the issue:
```
*************************************************
Checking External Tool Configuration (https://terostechnology.github.io/terosHDLdoc/docs/external_tools/) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Selected external tool: MODELSIM. Installation path: "/home/volta/intelFPGA/20.1/modelsim_ase/linux"
  🔎 Searching for the binary modelsim at "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim"
  ✅ modelsim found at "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim" using the command: "/home/volta/intelFPGA/20.1/modelsim_ase/linux/modelsim --version"
🎉 The MODELSIM installation path is correctly configured.

⊙ Execution mode: CMD. The tool will be executend in the command line.

⊙ Waveform viewer: TOOL. Built-in tool waveform viewer will be opened after the simulation if it is available.
```

el3ctrician, 2025-01-22:
@qarlosalberto should be fixed also before #718  right ? can we close it ?

qarlosalberto, 2025-01-22:
Solved, the new version will be released to the market soon.

closed 2025-01-22 by qarlosalberto