TerosTechnology/vscode-terosHDL #707: Export settings button creates file with contents "[object Object]"

sirnails, 2024-12-06:
**Describe the bug**
Clicking the "Export Settings" button in TerosHDL creates a file, but the file's content is simply [object Object]. This suggests the settings object is not being serialized correctly before being written to the file.

**To Reproduce**
Steps to reproduce the behavior:
1. Open TerosHDL in VSCode.
2. Scroll down to configuration.
3. Click on the "Export Settings" button.
4. Navigate to desktop folder
5. Type teros.json
6. Inspect the contents of the generated file.

**Code**
Code to reproduce the error.

**Please complete the following information:**
 - OS: Microsoft Windows [Version 10.0.19045.5131]
 - VSCode version version
 - Version: 1.95.3 (user setup)
Commit: f1a4fb101478ce6ec82fe9627c43efbf9e98c813
Date: 2024-11-13T14:50:04.152Z
Electron: 32.2.1
ElectronBuildId: 10427718
Chromium: 128.0.6613.186
Node.js: 20.18.0
V8: 12.8.374.38-electron.0
OS: Windows_NT x64 10.0.19045

**Screenshots**
Not applicable, as the issue is with file contents.

**Additional context**
On an tangent, Its not clear to me if this is intended to export the global settings or the project settings.

el3ctrician, 2025-02-12:
Same under linux, i will have a look when i have time

jevogel, 2025-02-28:
> Its not clear to me if this is intended to export the global settings or the project settings.

I recently found out that you can export project settings by going to the top of the Files pane in TerosHDL and clicking the Save Project 💾 icon. You can then import this project file using the "Load project from YAML EDAM" option, though it will silently fail if there is an existing project with the same name.

So, I think the Export Settings button is intended to export the global settings. My colleague told me that this used to work properly before, but that was back when there was only one global configuration and no individual project configuration, so the bug might have been introduced when per-project configuration was added.

qarlosalberto, 2025-03-02:
@jevogel could you send me an email?

qarlosalberto, 2025-03-18:
Fixed in v7.0.0. Feel free to reopen the issue if it doesn't work as expected.

closed 2025-03-18 by qarlosalberto