TerosTechnology/vscode-terosHDL #710: "Copy as " tool always open console

wimille, 2024-12-20:
Hi

I don't know if it is a bug of TerosHDL, an unknown option i've missed, or even a bug in my vscode configuration. So i create a blank issue.

When i open any VHDL file, select the entity and click on "Generate Template / Copy as signal", each time vscode open the output console "TerosHDL: Global". It is kind of boring. If there is error ok, but juste to say "copy" it is not interesting. 

Thanks

el3ctrician, 2025-01-31:
@qarlosalberto i think that there is no need to call the globalLogger [here](https://github.com/TerosTechnology/vscode-terosHDL/blob/4f8a47aa3155b1f7c7ba4d14c6ceee300415f808/src/teroshdl/features/templates.ts#L82) with enable show true, maybe set it to false or call the toolLogger to avoid having the terminal to pop-up when it's not needed ?

qarlosalberto, 2025-01-31:
Maybe it's better:

- If it fails open the console (as now)
- If it works fine to show a message with: `vscode.showInformationMessage('.....');`

wimille, 2025-02-04:
Hi both

I think it is a great idea to open console only if copy failed. Hope this feature will be include in future release.

qarlosalberto, 2025-02-04:
@el3ctrician can you do it?

el3ctrician, 2025-02-08:
fixed, this can be close

closed 2025-02-08 by qarlosalberto