TerosTechnology/vscode-terosHDL #714: Can't get vivado to work as linter

nema69, 2025-01-15:
**Describe the bug**
Teros doesn't find vivado when it's configured as the linter. Verify setup claims it found "vivado" in the system path when checking the external tool configuration, but when checking linter configuration it doesn't find "Vivado" (see screenshot). Is the different capitalization on purpose? Also when I click on "Verify Setup" Vivado opens. 

PS: The command "TerosHDL:Verify Setup" returns "command 'teroshdl.configuration.verifySetup' not found" , but the button seems to work. 

**To Reproduce**
Choose Vivado(xvhdl) as the linter in Linter settings. Provide the path to the vivado installation folder in Tools->Vivado->Installation path

**Please complete the following information:**
 - OS: Arch Linux
 - VSCode version version v6.0.14

**Screenshots**
[image]

Thanks

nema69, 2025-01-16:
Got a bit further:
1. Originally I was editing the global settings while a project was open. This didn't do anything as the project settings overwrote the global settings in this case. It would be convinient to have an indication that a global setting is being overwritten by the project settings.
2. Vivado is not found even when the correct path is used. The ouput of verify setup is:

`🔎 Searching for the binary vivado at "/home/lr/vivado/opt/Xilinx/Vivado/2023.2/bin/vivado"
  ❌ vivado not found at "/home/lr/vivado/opt/Xilinx/Vivado/2023.2/bin/vivado". Search executed with: "/home/user/vivado/opt/Xilinx/Vivado/2023.2/bin/vivado --version"`

If I enter the same command in my terminal I get:

`[user@user_pc bin]$ /home/user/vivado/opt/Xilinx/Vivado/2023.2/bin/vivado --version
ERROR: [Common 17-170] Unknown option '--version', please type 'vivado -help' for usage info.
`
It seems to me that Teros is looking for vivado with the "--version" flag which is not a valid flag, but "-version" is.

qarlosalberto, 2025-01-22:
Solved, the new version will be released to the market soon.

closed 2025-01-22 by qarlosalberto