TerosTechnology/vscode-terosHDL #715: Vivado linting not detected

ht145, 2025-01-18:
[image]

currently, an issue where Teros is searching for "-version" instead of "vivado -version"'

vivado seems to be correctly added to windows path as seen here. 
[image]

el3ctrician, 2025-01-22:
duplicate from #701

closed 2025-01-22 by qarlosalberto

rloconnor, 2025-01-30:
Has this been tested on Windows 10? The extension attempts to find vivado.exe, but Vivado should be launched through vivado.bat on Windows. @el3ctrician

el3ctrician, 2025-01-30:
Are you using the latest from this repo ? I think the fix is not released yet in the new version. If so could you please share the whole log when checking for vivado ? I use linux and don't have a windows machine...
Thanks

rloconnor, 2025-01-31:
Thanks for the response, looks it's working now - I can see errors about things not being compiled into the work library in VSCode. 

I'm using the latest .vsix downloaded from this repo which includes this fix based on the changlog.

I hadn't set Vivado as the main tool on the tools settings page. However, the log is still showing that it can't find Vivado so I've included that here, along with the default configuration file. Interstingly I've got Vivado configured in my path, but it can't find it even when running vivdo -version.

[.teroshdl2_config.json](https://github.com/user-attachments/files/18614622/default.teroshdl2_config.json)

[teroshdl.log](https://github.com/user-attachments/files/18614641/teroshdl.log)

rloconnor, 2025-02-03:
@el3ctrician FYI, I'm not going to have time to do this until the weekend, will look into it then

rloconnor, 2025-02-09:
@el3ctrician 

I have run the settings64.bat file and that seems to fix the issue as expected by adding Vivado to the path (previously I had done this maually). However, the log still indicates that Vivado is not found (attached). It looks like the log output is incorrect because I am definitely getting errors highlighted by Vivado's linter in VSCode (see below). I have also manually configured the path to Vivado bin directory in settings (attached) but this does not make any difference to the log output.

[image]

[log.txt](https://github.com/user-attachments/files/18724212/log.txt)
[.teroshdl2_config.json](https://github.com/user-attachments/files/18724201/default.teroshdl2_config.json)

el3ctrician, 2025-02-12:
did you lunch setting64.bat before lunching teroshdl ? maybe the settings are applyed later ? in anycase i unfortunatlly can't debug it over windows in these day but i will try to run a VM and see what happens in windows

cezchitos, 2025-02-14:
I have the same problem, in that the Vivado linter cannot be be found on Windows.

[image]

If I manually run the command underlined in red, that works fine.    

[image]

Furthermore, I think there is also an issue with the External Tool Configuration, as that uses '--version' instead of '-version' (extra dash). 

[image]

Should I create a new bug for this or is this one going to be reopened?

rloconnor, 2025-02-15:
> did you lunch setting64.bat before lunching teroshdl ? maybe the settings are applyed later ? in anycase i unfortunatlly can't debug it over windows in these day but i will try to run a VM and see what happens in windows

Yes I did, I ran  settings64.bat and then ran the Verify Setup process

PR #718 (2025-01-22): fix vivado default binary name to be able to correctly check for it
that should fix #715 #714 #701