TerosTechnology/vscode-terosHDL #717: Schematic Viewer fails (GHDL + Yosys, Windows)

gfcwfzkm, 2025-01-20:
**Describe the bug**
The schematic viewer fails on windows for VHDL (GHDL + YOSYS):
```bash
2025-01-20 11:17:17.731 [info] Executing: C:\Windows\system32\cmd.exe  C:\Windows\system32\cmd.exe /d /s /c " C:\msys64\ucrt64\bin\yosys.exe -m ghdl -p "ghdl --std=08 -fsynopsys  --work=work "c:\Users\WDAGUtilityAccount\.vscode\extensions\teros-technology.teroshdl-6.0.14\resources\project_manager\examples\ghdl\half_adder.vhd"  --work=work -e half_adder; hierarchy -top half_adder; proc; write_json C:\Users\WDAGUtilityAccount\.teroshdl_8RZk0; stat""
2025-01-20 11:17:17.963 [info] ERROR: Can't load module `./ghdl': "C:\msys64\ucrt64\bin\..\share\yosys\plugins\ghdl.so": The specified module could not be found.
2025-01-20 11:17:17.971 [error] Yosys failed.
```

The bug report also applies when trying to execute the schematic viewer from the hierarchy.

**To Reproduce**
GHDL and Yosys (as well as GNU Make and GTKWave) were installed using MSYS2 UCRT64 with the following command:
```bash
pacman -S make mingw-w64-ucrt-x86_64-make mingw-w64-ucrt-x86_64-ghdl-llvm mingw-w64-ucrt-x86_64-gtkwave mingw-w64-ucrt-x86_64-yosys
```
The binary paths to Yosys and GHDL are both set to `C:\msys64\ucrt64\bin` and performing the "Verify Setup" task successfully detects them.

Is this a good way to install GHDL + Yosys on Windows? If not, what would be better? I'm using MSYS2 to benefit from an simple update system via pacman but if there is a more reliable way to use GHDL+Yosys with TerosHDL, I'm open for it.

**Code**
The error can be reproduced with the GHDL example included by TerosHDL.

**Please complete the following information:**
 - OS: Windows 11 Enterprise *and* Windows 11 Pro
 - VSCode version: 1.96.4
 - TerosHDL version: 6.0.14

**Additional context**
If I perform the command in CMD manually, but remove `-m ghdl` from the command, it seems that everything performs as expected and a JSON file is written.
```bash
C:\Windows\system32\cmd.exe /d /s /c " C:\msys64\ucrt64\bin\yosys.exe -p "ghdl --std=08 -fsynopsys  --work=work "c:\Users\WDAGUtilityAccount\.vscode\extensions\teros-technology.teroshdl-6.0.14\resources\project_manager\examples\ghdl\half_adder.vhd"  --work=work -e half_adder; hierarchy -top half_adder; proc; write_json C:\Users\WDAGUtilityAccount\.teroshdl_8RZk0; stat""
```
It seems that the MSYS2 versions of Yosys and GHDL don't ship GHDL as shared library (there is no `\share\yosys` folder either), yet executing `ghdl` within Yosys performs just fine.

JakubFranek, 2025-02-16:
Hello

I have the same issue.

I think the problem is a bug in TerosHDL. The MSYS2 version of yosys, (package `mingw-w64-yosys` to be exact), already contains `ghdl-yosys-plugin` built into yosys ([see this file for proof](https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-yosys/PKGBUILD)).

Therefore installing yosys this way does not create any `plugins/ghdl.so` file. The ghdl integration is now baked into yosys and creating the schematic diagram should be invoked somehow differently by the TerosHDL extension. I do not know how, but as of now the schematic viewer on Windows is not usable. 

I will be happy to learn any workaround. Thank you!

Jakub

qarlosalberto, 2025-02-16:
Could you share an example?

JakubFranek, 2025-02-16:
Hello @qarlosalberto, what kind of example do you have in mind?

I installed ghdl and yosys (with a built-in ghdl-yosys-plugin) via MSYS2 from the [mingw-w64-x86_64-eda](https://packages.msys2.org/group/mingw-w64-x86_64-eda) package group. This installation creates working yosys and ghdl, but it does not create any ghdl.so file, which TerosHDL expects. This creates the following error when attempting to view a schematic diagram of a VHDL code within TersosHDL:

> 2025-02-16 21:06:06.483 [info] Executing: C:\WINDOWS\system32\cmd.exe  C:\WINDOWS\system32\cmd.exe /d /s /c " D:\Programy\MSYS2\mingw64\bin\yosys.exe -m ghdl -p "ghdl --std=08 -fsynopsys  --work=work "d:\Electronics\FPGA\nexys-a7-lab\source\common\clock_enable_generator.vhd"  --work=work -e clock_enable_generator; hierarchy -top clock_enable_generator; proc; write_json C:\Users\jfran\.teroshdl_wfim2; stat""
> 2025-02-16 21:06:06.548 [info] ERROR: Can't load module `./ghdl': "D:\Programy\MSYS2\mingw64\bin\..\share\yosys\plugins\ghdl.so": Uveden� modul nebyl nalezen.
> 
> 2025-02-16 21:06:06.550 [info] 
>  /----------------------------------------------------------------------------\
>  |                                                                            |
>  |  yosys -- Yosys Open SYnthesis Suite                                       |
>  |                                                                            |
>  |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
>  |                                                                            |
>  |  Permission to use, copy, modify, and/or distribute this software for any  |
>  |  purpose with or without fee is hereby granted, provided that the above    |
>  |  copyright notice and this permission notice appear in all copies.         |
>  |                                                                            |
>  |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
>  |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
>  |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
>  |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
>  |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
>  |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
>  |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
>  |                                                                            |
>  \----------------------------------------------------------------------------/
> 
>  Yosys 0.38 (git sha1 543faed9c8c, x86_64-w64-mingw32-g++ 13.2.0 -march=nocona -msahf -mtune=generic -O2 -fstack-protector-strong -Os)
> 
> 
> 2025-02-16 21:06:06.555 [error] Yosys failed.

The original poster of this issues says that removing `-m ghdl` from the command fixed the issue for him, so I tried it. 

I had to clean up the command a little bit into this form first: 
`D:/Programy/MSYS2/mingw64/bin/yosys.exe -p "ghdl --std=08 -fsynopsys --work=work d:/Electronics/FPGA/nexys-a7-lab/source/common/clock_enable_generator.vhd --work=work -e clock_enable_generator; hierarchy -top clock_enable_generator; proc; write_json C:/Users/jfran/.teroshdl_gyYxC; stat"
`

When I ran it, I got this:

> jfran@Lenovo-Y530 MINGW64 /d/Electronics/FPGA/nexys-a7-lab (master)
> $ D:/Programy/MSYS2/mingw64/bin/yosys.exe -p "ghdl --std=08 -fsynopsys --work=work d:/Electronics/FPGA/nexys-a7-lab/source/common/clock_enable_generator.vhd --work=work -e clock_enable_generator; hierarchy -top clock_enable_generator; proc; write_json C:/Users/jfran/.teroshdl_gyYxC; stat"
> 
>  /----------------------------------------------------------------------------\
>  |                                                                            |
>  |  yosys -- Yosys Open SYnthesis Suite                                       |
>  |                                                                            |
>  |  Copyright (C) 2012 - 2020  Claire Xenia Wolf <claire@yosyshq.com>         |
>  |                                                                            |
>  |  Permission to use, copy, modify, and/or distribute this software for any  |
>  |  purpose with or without fee is hereby granted, provided that the above    |
>  |  copyright notice and this permission notice appear in all copies.         |
>  |                                                                            |
>  |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
>  |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
>  |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
>  |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
>  |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
>  |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
>  |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
>  |                                                                            |
>  \----------------------------------------------------------------------------/
> 
>  Yosys 0.38 (git sha1 543faed9c8c, x86_64-w64-mingw32-g++ 13.2.0 -march=nocona -msahf -mtune=generic -O2 -fstack-protector-strong -Os)
> 
> 
[... cut ...]
> Yosys 0.38 (git sha1 543faed9c8c, x86_64-w64-mingw32-g++ 13.2.0 -march=nocona -msahf -mtune=generic -O2 -fstack-protector-strong -Os)
> Time spent: 6% 2x write_json (0 sec), 6% 2x proc_clean (0 sec), ...

It seems that the operation was successful. Therefore **I would propose to create another GHDL+Yosys option in the TerosHDL Global Settings/Schematic viewer/Backend, which would send nearly the same command as it does now, but without the `-m ghdl`.**

Hope it makes sense and hope it works for others as well. Please let me know if you need any further information.

Kind regards
Jakub

JakubFranek, 2025-02-21:
Hello @qarlosalberto 

I traced the issue in the source code of TerosHDL to [src/colibri/yosys/yosys.ts, line 154](https://github.com/TerosTechnology/vscode-terosHDL/blob/dev/src/colibri/yosys/yosys.ts#L154).

```
let cmd =
        // eslint-disable-next-line max-len
        `${preArguments} ${yosysPath} -m ghdl -p "ghdl --std=08 -fsynopsys ${ghdlArguments} ${cmdFiles} --work=work -e ${topTevel}; ${topLevelCmd}; proc; ${customArguments}; write_json ${outputPathFilename}; stat"`;
    cmd = removeEmptyCommands(cmd);
```

The issue is the `-m ghdl` right after `${yosysPath}`. I believe both me and @gfcwfzkm demonstrated that removing `-m ghdl` would fix the issue for users who installed `yosys` from the MSYS2 mingw package. On the other hand, it would like break this for people, whose setup includes the `ghdl.so` file separately as a plugin. 

I think a setting allowing the user to specify whether he wants the `-m ghdl` there or not would be ideal. Default behaviour could stay the same and this fix would therefore have no effect for other users.

I would be happy to help testing any potential fixes to this issue if desired.

Kind regards
Jakub

qarlosalberto, 2025-02-22:
I had the option, but I removed it. I would like to ro the configuration simplest as possible. 
I think that the best way would be to run with -m and if it fails to remove -m and re run

JakubFranek, 2025-02-23:
Hello

just in case anyone else has the same issue and wants a quick fix, they might find [this Python script](https://github.com/JakubFranek/Nexys-A7-Lab/blob/master/scripts/synthesize_svg.py) I created helpful. It requires yosys and netlistsvg, and creates a synthesized SVG schematic of the input .vhd file. Please note you will need to change the paths in the script before running it, they are listed at the top of the file just beneath the imports. You can run it like this: `python synthesize_svg.py my_module.vhd my_module.svg`.

Regards
Jakub

closed 2025-03-18 by qarlosalberto