TerosTechnology/vscode-terosHDL #721: Add following 'ifdef - 'endif bracket pairs

niciki-niciki, 2025-02-05:
I have a ready solution to apply (below).
This simly add another brecket pairs:
. 'ifdef - 'endif
. 'ifndef - 'endif

In `./teros-technology.teroshdl-6.0.14/configs` edit two files:
. systemverilog.configuration.json
. verilog.configuration.json

Please add following brackets pairs:
```JSON
"brackets": [
...
  ["`ifdef", "`endif"],
  ["`ifndef", "`endif"]
],
```

And please add following autoClosingPairs:
```JSON
"autoClosingPairs": [
...
  {"open":"`ifdef", "close":"`endif"},
  {"open":"`ifndef", "close":"`endif"}
],
```

Thanks,
Daniel

qarlosalberto, 2025-02-05:
Could you create a pull request? https://github.com/TerosTechnology/vscode-terosHDL/tree/dev/configs

niciki-niciki, 2025-02-05:
This will be my first time.
I will try and learn the process.

qarlosalberto, 2025-03-18:
Added. The change will be available in the next release.

closed 2025-03-18 by qarlosalberto