TerosTechnology/vscode-terosHDL #725: Documenter produces incorrectly formatted markdown tables from comments

brandonhippe, 2025-02-17:
**Describe the bug**
Markdown tables included for auto-documentation are correctly formatted in the documentation preview, but lose newlines after export, causing tables to not render properly.

**To Reproduce**
Export as markdown using [https://github.com/TerosTechnology/teroshdl-examples/blob/main/documenter/examples/markdown_example.vhd](markdown_example.vhd) as input

**Code**
```
-- This file is public domain, it can be freely copied without restrictions.
-- SPDX-License-Identifier: CC0-1.0
library ieee;
  use ieee.std_logic_1164.all;
  use ieee.numeric_std.all;

--! This example uses as symbol to extract the comment "!". It
--! can be configured in TerosHDL.
--! You can use Markdown github flavour tags. A table example:
--!
--! | t0 | table | example |
--! |----|-------|---------|
--! | 1  | 10    | 9       |
--! | 2  | 4     | 5       |
--! | 33 | 12    | 22      |
--!
--! Other table:
--!
--! | t0 | fix | example |
--! |----|-----|---------|
--! | 1  | 10  | 9       |
--! | 2  | 4   | 5       |
--! | 33 | 12  | 22      |

entity adder is
  generic (

    DATA_WID : positive := 4);
  port (
    A : in    unsigned(DATA_WID-1 downto 0); --! A input
    B : in    unsigned(DATA_WID-1 downto 0); --! B input
    X : out   unsigned(DATA_WID downto 0) --! Output adder
  );  
end entity adder;

architecture rtl of adder is
  constant cnt_SAMPLE_0 : integer := 0;

  --! Example **bold description in** constant
  constant cnt_SAMPLE_1 : integer := 0;
  
  signal signal_example : std_logic;
begin
  signal_example <= '0';

  --! Description in process
  add_proc : process (A, B) is
  begin
    X <= resize(A, X'length) + B;
  end process add_proc;

  --! Instantiation description example
  --! with **markdown bold**
  MUX_1 : mux
    generic map (n => 2)
    port map (
      sel => '1',
      din => '1',
      q => '0'
    );

end architecture rtl;
```

**Please complete the following information:**
 - OS: Ubuntu 24.04
 - VSCode version 1.97.0

**Screenshots**
Output markdown:
```

# Entity: adder 
- **File**: markdown_example.vhd

## Diagram
![Diagram](adder.svg "Diagram")
## Description

This example uses as symbol to extract the comment "!". It can be configured in TerosHDL. You can use Markdown github flavour tags. A table example:

| t0 | table | example | |----|-------|---------| | 1  | 10    | 9       | | 2  | 4     | 5       | | 33 | 12    | 22      |

Other table:

| t0 | fix | example | |----|-----|---------| | 1  | 10  | 9       | | 2  | 4   | 5       | | 33 | 12  | 22      | 
## Generics

| Generic name | Type     | Value | Description |
| ------------ | -------- | ----- | ----------- |
| DATA_WID     | positive | 4     |             |

## Ports

| Port name | Direction | Type                          | Description  |
| --------- | --------- | ----------------------------- | ------------ |
| A         | in        | unsigned(DATA_WID-1 downto 0) | A input      |
| B         | in        | unsigned(DATA_WID-1 downto 0) | B input      |
| X         | out       | unsigned(DATA_WID downto 0)   | Output adder |

## Constants

| Name         | Type    | Value | Description                              |
| ------------ | ------- | ----- | ---------------------------------------- |
| cnt_SAMPLE_1 | integer | 0     | Example **bold description in** constant |

## Processes
- add_proc: ( A, B )
  - **Description**
  Description in process

## Instantiations

- MUX_1: mux
  -  Instantiation description example with **markdown bold**
```

**Additional context**
Add any other context about the problem here.

qarlosalberto, 2025-03-18:
Fixed in v7.0.0. Feel free to reopen the issue if it doesn't work as expected.

brandonhippe, 2025-03-18:
Same behavior in v7.0.0

qarlosalberto, 2025-03-19:
I didn't push the changes. Check the new v7.0.1

closed 2025-03-19 by qarlosalberto

brandonhippe, 2025-03-25:
Works now, thanks!