TerosTechnology/vscode-terosHDL #726: vhdl-ls processes remain open even after closing VSCode

mth201698, 2025-02-19:
**Describe the bug**
The vhdl-ls process remains open even after my instances of VSCode is closed.

**To Reproduce**
I use the VSCode ssh extension to develop on a server. The server does not usually get rebooted, which is why I am able to notice this issue. If it goes on for long enough, it actually maxes out the allowable RAM usage for my account and I have to manually kill the processes to be able to run anything on my account. I have not observed this issue when running VSCode with the teroshdl extension locally, only when running on the server via ssh.

**Please complete the following information:**
 - OS: Red Hat Enterprise Linux Server 7.9
 - TerosHDL version: 6.0.14
 - VSCode version: 
                Version: 1.97.2 (user setup)
                Commit: e54c774e0add60467559eb0d1e229c6452cf8447
                Date: 2025-02-12T23:20:35.343Z
                Electron: 32.2.7
                ElectronBuildId: 10982180
                Chromium: 128.0.6613.186
                Node.js: 20.18.1
                V8: 12.8.374.38-electron.0
                OS: Windows_NT x64 10.0.19045

matthias-vorwerk, 2025-03-06:
I see the same problem. As we have a multi-user system this results in remaining running processes after existing vscode.

qarlosalberto, 2025-03-18:
I have added a timeout in the v7.0.0. Let see if it works.
Feel free to reopen the issue if it doesn't work as expected.