TerosTechnology/vscode-terosHDL #729: Unable to set up VUnit

JakubFranek, 2025-02-23:
**Describe the bug**
I am having trouble setting up VUnit with TerosHDL. I have a run.py script, which contains this:

```
from vunit import VUnit

vu = VUnit.from_argv(compile_builtins=False)
vu.add_vhdl_builtins()
lib = vu.add_library("lib")
lib.add_source_files("source/**/*.vhd")
lib.add_source_files("testbench/**/*.vhd")
vu.main()
```

I also have a testbench which looks like this:

```
library vunit_lib;
context vunit_lib.vunit_context;

entity tb_example is
    generic (runner_cfg : string);
end entity;

architecture tb of tb_example is
begin
    main : process
    begin
        test_runner_setup(runner, runner_cfg);
        report "Hello world!";
        test_runner_cleanup(runner); -- Simulation ends here
    end process;
end architecture;
```

When I run python run.py, VUnit runs correctly.

What I would like is to be able to run this from TerosHDL, or at least to be able to correctly lint the test bench files. But as of now, the built-in VHDL linter does not recognize 

```
No such library 'vunit_lib'
No declaration of 'vunit_lib'
No declaration of 'test_runner_setup'
No declaration of 'runner'
```

**To Reproduce**
Code to reproduce the error.

**Code**
Code to reproduce the error.

 - OS: Windows 10
 - VSCode version 1.97.2

**Screenshots**
If applicable, add screenshots to help explain your problem. **Share the code as text, not as a screenshots!**

**Additional context**
Add any other context about the problem here.

qarlosalberto, 2025-02-23:
Currently in TerosHDL you can't do both things in the same time (lint VUnit + run the run.py). If you want to lint your VUnit file libraries you can use the `watchers`: https://terostechnology.github.io/terosHDLdoc/docs/project_configuration/watcher

JakubFranek, 2025-02-23:
Hello @qarlosalberto 

sorry I accidentally hit Enter too fast and posted the issue before completing the description. Please take another look, I edited it.

Thank you for the advice. If I must choose I would prefer to be able to sucessfully lint my testbenches rather than run them (I can always run them via script). 

**I tried to add `run.py` to the watcher list as seen in the screenshot in my edited post, but the linter errors remain.**

I have the following questions:
1) Do I have to manually add `vunit_lib` to Files view, and if yes, how? I think `vunit_lib` consists of many .vhd source files in many sub-folders within my `.venv/Lib/site-packages/vunit directory`, it would be very difficult to add them all, but I could do it if this is the correct way, but it would be really exhausting.
2) Do I have to specify `vunit` path in TerosHDL config? I tried having empty Installation path for `vunit` as well as supplying path to the `.venv/Lib/site-packages/vunit directory`, but none of the options worked.

[image]

FYI when I do Verify Setup, I get the following output:

```
2025-02-23 12:05:23.177 [info] 
*************************************************
Checking general configuration (https://terostechnology.github.io/terosHDLdoc/docs/category/installation-checklist) 
❌ Some checks failed ❌
*************************************************
⊙ Checking make. This tool is used to run some external tools. Current configured installation path: "D:\Programy\MSYS2\usr\bin"
  🔎 Searching for the binary make at "D:\Programy\MSYS2\usr\bin\make.exe"
  ✅ make found at "D:\Programy\MSYS2\usr\bin\make.exe" using the command: "D:\Programy\MSYS2\usr\bin\make.exe --version"
🎉 The make installation path is correctly configured.

⊙ Checking Python installation. Configured installation path: "D:\Electronics\FPGA\nexys-a7-lab\.venv\Scripts\python.exe"
  ✅ Python was found in the path "D:\Electronics\FPGA\nexys-a7-lab\.venv\Scripts\python.exe". Tried to find it using the command: "D:\Electronics\FPGA\nexys-a7-lab\.venv\Scripts\python.exe -c "import sys; check_version = sys.version_info >         (3,0); exit(0) if check_version == True else exit(-1)""
🎉 Python installation path is correctly configured. Using "D:\Electronics\FPGA\nexys-a7-lab\.venv\Scripts\python.exe"

⊙ Checking Python dependencies. Current configured installation path: "D:\Electronics\FPGA\nexys-a7-lab\.venv\Scripts\python.exe"
  🎉 vunit found.
  🎉 edalize found.
  🎉 cocotb (optional) found.
  🎉 vsg (optional) found.
*************************************************
Checking Linter configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/linter) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Linter errors for VHDL is disabled. Skipping configuration check.

⊙ Linter errors for Verilog/SV is disabled. Skipping configuration check.

⊙ Linter style for VHDL is disabled. Skipping configuration check.

⊙ Linter style for Verilog/SV is disabled. Skipping configuration check.

*************************************************
Checking External Tool Configuration (https://terostechnology.github.io/terosHDLdoc/docs/external_tools/) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Selected external tool: GHDL. Installation path: "D:\Programy\MSYS2\mingw64\bin"
  🔎 Searching for the binary ghdl at "D:\Programy\MSYS2\mingw64\bin\ghdl.exe"
  ✅ ghdl found at "D:\Programy\MSYS2\mingw64\bin\ghdl.exe" using the command: "D:\Programy\MSYS2\mingw64\bin\ghdl.exe --version"
🎉 The GHDL installation path is correctly configured.

⊙ Execution mode: CMD. The tool will be executend in the command line.

⊙ Waveform viewer: TOOL. Built-in tool waveform viewer will be opened after the simulation if it is available.

*************************************************
Checking Formatter configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/formatter) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Checking formatter standalone for VHDL.
  The formatter is built into TerosHDL. Skipping configuration check.
🎉 The linter installation path is correctly configured.

⊙ Checking formatter istyle for Verilog/SV.
  The formatter is built into TerosHDL. Skipping configuration check.
🎉 The linter installation path is correctly configured.

*************************************************
Checking Schematic configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/schematic_viewer/installation) 
🎉 Correctly configured 🎉🎉
*************************************************
⊙ Selected backend: yosys_ghdl with installation path: "D:\Programy\MSYS2\mingw64\bin". 
  🔎 Searching for the binary yosys_ghdl at "D:\Programy\MSYS2\mingw64\bin\yosys.exe"
  ✅ yosys_ghdl found at "D:\Programy\MSYS2\mingw64\bin\yosys.exe" using the command: "D:\Programy\MSYS2\mingw64\bin\yosys.exe --version"
🎉 The Schematic Backend installation path is correctly configured.

```

So I think `vunit` is found correctly by TerosHDL. I must be missing something...

JakubFranek, 2025-02-23:
Hello @qarlosalberto,

FYI I managed to "fix" the linter errors by manually creating a `vhdl_ls.toml` file within my repository root, where I included the `vunit_lib` path into my Python .venv. This created its own set of problems, but those are likely outside of TerosHDL scope ([see this issue](https://github.com/VHDL-LS/rust_hdl/issues/366)).

However I am still perplexed why the TerosHDL-VUnit integration does not work for me. This is what I tried:
A) Create Project - Load project from VUnit run.py 
-- nothing happens at all, no Files are added, no Watcher is loaded, nothing is written into the TerosHDL output logs so no idea what went wrong
B) Create Project - Create an empty generic project - Add Source - Load from VUnit run.py
-- same as above, nothing happens at all
C) Create Project - Create an empty generic project - Add Watcher - VUnit - run.py
-- `run.py` appears in Watchers view, but Files view stays empty

My expectation was that all of these methods, but hopefully at least one, would automatically add all source files that match the following two lines in my `run.py` into the TerosHDL Files view:
```
lib.add_source_files("source/**/*.vhd")
lib.add_source_files("testbench/**/*.vhd")
```

jevogel, 2025-02-28:
Hi @JakubFranek,

I've been doing some experimentation with my Python environment and TerosHDL and come to the following conclusions.

TerosHDL does not correctly implement support for Python virtual environments. When the Verify Setup routine runs, it first searches for a Python environment, and then it searches for the four dependencies inside that environment. However, when it actually makes use of those dependencies, it does so in the system environment.

Try installing VUnit and VSG in your system environment like `pip install vunit-hdl vsg --break-system-packages` (on Ubuntu 24.04 LTS) without changing the Python path in your TerosHDL configuration and see if that works.

JakubFranek, 2025-03-01:
Hi @jevogel, thanks for your input. 
I just checked and I actually installed `vunit-hdl` into both my venv and my system environment. 
I think there must be some other reason for why I cannot set up vunit with TerosHDL.

qarlosalberto, 2025-03-18:
For VSG you need to add it to the System Path. But for VUnit you can configure your Virtual Env in Settings->General->Python3 Binary Path: https://terostechnology.github.io/terosHDLdoc/docs/installation_checklist/installation#2-python3