TerosTechnology/vscode-terosHDL #732: Allow libraries to be marked as "third-party" to disable VHDL-LS linting and VSG style checking

jevogel, 2025-02-28:
**Is your feature request related to a problem? Please describe.**

I use VUnit with VHDL. I want to use VHDL-LS and VSG on my VHDL source files and my VHDL testbench files. My VHDL testbench files use components from VUnit, so I need to include them in my TerosHDL file list so that they are analyzed by VHDL-LS and it doesn't show errors where they are referenced. However, adding these files to my TerosHDL project results in the VS Code Problems pane filling up with hundreds of warnings about unused signals from VHDL-LS and style checks from VSG for files that I have no control over and would like to ignore. With all of this noise, the Problems pane is too cluttered to use for browsing for issues in my own codebase, and analyzing so many files slows VSG down.

**Describe the solution you'd like**

VHDL-LS supports marking each library as `is_third_party` in the `vhdl_ls.toml` file, as in the following example:

```toml
[libraries]
vunit_lib.files = [ 'vunit/vhdl/**/*.vhd' ]
vunit_lib.is_third_party = true
```

TerosHDL could add a toggle option next to each library to mark it as a third-party dependency, and this could then just be passed directly into the generated `~/.vhdl_ls.toml` file created by the project. Additionally, TerosHDL could ignore any files in libraries marked as third-party when generating VSG linter commands.

**Describe alternatives you've considered**

- Uninstalling TerosHDL and just using VHDL-LS and VSG on their own.
- Hiding the Problems pane and just ignoring it, only looking at the squiggles in the file I have open.

**Additional context**

[image]

jevogel, 2025-02-28:
My colleagues and I use TerosHDL every day in our jobs and we love it. Thanks for all of your hard work! Let me know if there's any way we can help improve the tool.

qarlosalberto, 2025-03-18:
Fixed in v7.0.0. Feel free to reopen the issue if it doesn't work as expected.

closed 2025-03-18 by qarlosalberto

rameloni, 2025-03-19:
Hi, I manage my teros project by automatically generating a yaml file. Then, I upload this file to vscode manually to load the project. 
I'm facing the problem of declaring third party libraries such as `xpm` or `unisim`, do you have an example or any suggestion on how I can reach this?

qarlosalberto, 2025-03-19:
TerosHDL uses this in the backend: https://github.com/VHDL-LS/rust_hdl?tab=readme-ov-file#configuration

rameloni, 2025-03-19:
> TerosHDL uses this in the backend: https://github.com/VHDL-LS/rust_hdl?tab=readme-ov-file#configuration

So, I can edit the `.vhdl_ls.toml` file to apply updates, right? However, after a change the project doesn't refresh automatically. How can I refresh the project from vscode or cli to apply the changes?

qarlosalberto, 2025-03-19:
No, you can't. TerosHDL manages the `.vhdl_ls.toml` automatically. If you do the changes manually it will be overwritten. You have to add the XMP files manually in the Project Manager GUI.

rameloni, 2025-03-19:
Alright, thanks because my problem is that I would want to ignore completely third party libraries (either VHDL or Verilog) and Xilinx libraries https://github.com/TerosTechnology/vscode-terosHDL/issues/688

EsbenMadsen, 2025-03-27:
I'm just jumping in here to say that the solution might be similar to what I requested and had fulfilled last week (#736), if its simply a matter of appending 

>library<.files = [ ]  (or maybe >library<.files = [blackbox.vhd] to not have an empty library?)
>library<.is_third_party = true

to the vhdl_ls.toml file. Although in my testing this hasn't stopped VHDL-ls from giving errors, but I can't say why because I don't know how is_third_party works. Maybe if you can get it to work by manually editing the .toml file, the above approach can be reused?

On a sidenote I'd be happy to add to the documentation as requested in #736 , but I don't know how (:

qarlosalberto, 2025-03-28:
@EsbenMadsen this is the documentation repository: https://github.com/TerosTechnology/terosHDLdoc

You only need to clone and modify the files in: https://github.com/TerosTechnology/terosHDLdoc/tree/main/docs They are markdown.

If you want to check the changes in local: https://github.com/TerosTechnology/terosHDLdoc?tab=readme-ov-file#teroshdl-documentation.

It could be nice if you add in the documentation the steps that you do to change the documentation. :D