TerosTechnology/vscode-terosHDL #736: VHDL-LS (RUST_HDL) configurations are overwritten by TerosHDL

EsbenMadsen, 2025-03-13:
**Describe the bug**
I need to change the VHDL-LS standard from 2008 to 1993 because of this issue https://github.com/VHDL-LS/rust_hdl/issues/153 , which has luckily already been solved in https://github.com/VHDL-LS/rust_hdl/pull/284 and solves the former issue. The issue when using TerosHDL is that it overwrites the configurations written in /user/>user</.vhdl_ls.toml when the Teros library in VScode is updated, thus undoing any changes made. So for now I have to manually set the VHDL-LS standard every time I update my Teros library.

**To Reproduce**
Go to the /user/>user</.vhdl_ls.toml file and delete a library or write some text, the Teros GUI in VScode will not change, but upon modifying the library the changes to /user/>user</.vhdl_ls.toml are overwritten by the changes made in the Teros library.

**Code**
See issue https://github.com/VHDL-LS/rust_hdl/issues/153.

**Please complete the following information:**
 - OS: Fedora 40
 - VSCode version 1.97.2

**Screenshots**
See issues https://github.com/VHDL-LS/rust_hdl/issues/153 and https://github.com/VHDL-LS/rust_hdl/pull/284

**Additional context**
The way I see it the communication between Teros and the vhdl_ls.toml file should be bidirectional, which would also make it possible to add/delete libraries in Teros using the vhdl_ls.toml file, which has applications when handling large projects. Alternatively (or additionally), adding a setting the allows you to set the standard used in VHDL-LS by adding "standard = "2008/93/etc."" to the top of the /user/>user</.vhdl_ls.toml in a currently non-existing settings page for VHDL-LS (or maybe the page for linter settings).

EsbenMadsen, 2025-03-17:
In hindsight this should have been a feature request, sorry about that. I decided to implement this myself and I believe I have figured out a solution or at least a "first step", but I don't know how to test it. I haven't contributed to a github project before, nor have I built a VScode extension from source. I followed the dev guide from the readme, but that didn't get me very far. A detailed explanation of how to test my code edits would be greatly appreciated!

qarlosalberto, 2025-03-18:
Done in v7.0.0. Feel free to reopen the issue if it doesn't work as expected.

closed 2025-03-18 by qarlosalberto