TerosTechnology/vscode-terosHDL #737: VSG does not work with YAML or JSON configuration file

JakubFranek, 2025-03-17:
**Describe the bug**
Without any YAML or JSON config file specified in TerosHDL Global Settings, VSG works fine as any rule violations within the examined .vhd file are highlighted and shown in the Problems section. As soon as a JSON or YAML config file is referenced, however, no rule violations are highlighted nor listed in Problems section.

**To Reproduce**
Enter a path to JSON or YAML file here:
[image]

**Code**
I tried both JSON and YAML config files, and I tried them both empty or with the following contents (for JSON):

```
{
    "rule":{
        "global":{
            "indent_size":4
        }
    }
}
```

In all cases, TerosHDL does not highlight or list any rule violations as soon as a config file is referenced in the settings.

**Please complete the following information:**
 - OS: Windows 10
 - VSCode version 1.97.2

**Additional context**
When the config file is referenced, the command in TerosHDL: Debug monitor looks correct, for example:
`vsg -f d:\Electronics\FPGA\Nexys-A7-Lab\source\clock_enable_generator\clock_enable_generator.vhd --all_phases -c vsg.json --js C:\Users\jfran\AppData\Local\Temp\f-2025217-11348-s0kkd3.nx7bn `

When I manually run this in a terminal, rule violations are listed in the terminal. So this does not seem to be an issue with vsg, but rather with TerosHDL when any config is used.

qarlosalberto, 2025-03-18:
Could you set the full path for `vsg.json` ?

I use this `vsg.yml` in TerosHDL and it works fine:

```
name : itcl_style
skip_phase: []
rule :
  global:
    case: 'lower'
    indent_size: 4
  group:
    indent:
      disable: false
  concurrent_003 :
    disable : false
  concurrent_009 :
    disable : false
    wrap_at_when: 'no'
    align_when_keywords: 'no'
    align_else_keywords: 'no'
  constant_012 :
    disable : false
  constant_014 :
    disable : false
  if_004 :
    disable : false
  process_020 :
    disable : true
  sequential_004 :
    disable : false
  variable_assignment_004 :
    disable : false
  whitespace_001:
    disable : true
  whitespace_002 :
    disable : false
  architecture_016:
    disable : true
  generic_007:
    disable : true
  architecture_015:
    disable : true
  package_003:
    sytle: 'require_blank_line'
  process_022:
    disable: true
  if_031:
    disable: true
  type_011:
    disable: true
  type_010:
    disable: true
  process_023:
    disable: true
  if_030:
    disable: true
  signal_007:
    disable: true
  instantiation_034:
    disable: true
  instantiation_004:
    disable: true
  generate_003:
    disable: true
  package_011:
    disable: true
  package_body_201:
    disable: true
  subprogram_body_201:
    disable: true
  subprogram_body_204:
    disable: true
  subprogram_body_202:
    disable: true
  subprogram_body_203:
    disable: true
  case_009:
    disable: true
  case_010:
    disable: true
  case_201:
    disable: true
  case_007:
    disable: true
  case_200:
    disable: true
  process_011:
    disable: true
  loop_statement_203:
    disable: true
  loop_statement_201:
    disable: true
  loop_statement_202:
    disable: true
  loop_statement_200:
    disable: true
  constant_015:
    disable: false
    prefixes: ['cnt_']
  constant_004:
    disable: false
    case: 'upper'
    prefix_exceptions: ['cnt_']

  generic_020:
    disable: false
    prefixes: ['g_']
  generic_007:
    disable: false
    case: 'upper'
    prefix_exceptions: ['g_']
  process_026:
    disable: true
  variable_007:
    disable: true

  port_010:
    case: 'lower'
  port_025:
    disable: false
    suffixes: ['_in', '_out']
    exceptions: ['clk', 'rst']
```

JakubFranek, 2025-03-19:
Hello @qarlosalberto,

thanks, putting the full absolute path seems to work. I thought relative paths would work and would use the VS Code's project folder as the current working directory. I should have realized this might be the fix though.

Thanks again and sorry for the trouble. Thanks for the config too!

Kind regards
Jakub

qarlosalberto, 2025-03-19:
Hmmm you are right I will add support for relative path.

qarlosalberto, 2025-03-20:
I'm thinking about this. It's not possible to work with relative paths. TerosHDL doesn't know your VSCode folder, it works with TerosHDL projects. So I close the issue.

closed 2025-03-20 by qarlosalberto