TerosTechnology/vscode-terosHDL #742: Error al abrir archivos System Verilog

Joworge, 2025-03-19:
Type: <b>Bug</b>

La primera línea se subraya en rojo y aparece el mensaje: "Execution of vlib.exe failed. Please check the error log for more details. TerosHDL: modelsim(v10g-66)".
No puedo compilar el código y Code Runner no reconoce el lenguaje.

Extension version: 7.0.1
VS Code version: Code 1.98.1 (2fc07b811f760549dab9be9d2bedd06c51dfcb9a, 2025-03-10T15:38:08.854Z)
OS version: Windows_NT x64 10.0.19045
Modes:

<details>
<summary>System Info</summary>

|Item|Value|
|---|---|
|CPUs|Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz (12 x 2208)|
|GPU Status|2d_canvas: enabled<br>canvas_oop_rasterization: enabled_on<br>direct_rendering_display_compositor: disabled_off_ok<br>gpu_compositing: enabled<br>multiple_raster_threads: enabled_on<br>opengl: enabled_on<br>rasterization: enabled<br>raw_draw: disabled_off_ok<br>skia_graphite: disabled_off<br>video_decode: enabled<br>video_encode: enabled<br>vulkan: disabled_off<br>webgl: enabled<br>webgl2: enabled<br>webgpu: enabled<br>webnn: disabled_off|
|Load (avg)|undefined|
|Memory (System)|15.85GB (7.17GB free)|
|Process Argv|C:\\Users\\pedra\\OneDrive\\Escritorio\\P3_2025_03_19 --crash-reporter-id 4cdffc02-395a-44fb-97f1-bf0f7e73a704|
|Screen Reader|no|
|VM|0%|
</details><details>
<summary>A/B Experiments</summary>

```
vsliv368:30146709
vspor879:30202332
vspor708:30202333
vspor363:30204092
vscod805cf:30301675
binariesv615:30325510
py29gd2263:31024239
c4g48928:30535728
azure-dev_surveyone:30548225
a9j8j154:30646983
962ge761:30959799
h48ei257:31000450
pythontbext0:30879054
cppperfnew:31000557
dwnewjupytercf:31046870
nativerepl1:31139838
pythonrstrctxt:31112756
nativeloc1:31192215
iacca1:31171482
5fd0e150:31155592
dwcopilot:31170013
6074i472:31201624
dwoutputs:31242946
customenabled:31248079
5a9hg449:31230677
hdaa2157:31222309
copilot_t_ci:31222730
f5992895:31254866
jda6j935:31233686
bgtreat:31256532
4d2d8976:31256800
hb9j0513:31256344

```

</details>

qarlosalberto, 2025-03-19:
> Type: *Bug*
> VM 0% A/B Experiments

A qué te refieres con que no puedes compilar y qué es code runner?

Joworge, 2025-03-19:
> A qué te refieres con que no puedes compilar y qué es code runner? <br>
> <br> &gt; Type: *Bug* <br> &gt; <br> &gt; La primera línea se subraya en
> rojo y aparece el mensaje: &quot;Execution of <br> &gt; vlib.exe failed.
> Please check the error log for more details. TerosHDL: <br> &gt;
> modelsim(v10g-66)&quot;. <br> &gt; No puedo compilar el código y Code
> Runner no reconoce el lenguaje. <br> &gt; <br> &gt; Extension version:
> 7.0.1 <br> &gt; VS Code version: Code 1.98.1
> (2fc07b811f760549dab9be9d2bedd06c51dfcb9a, <br> &gt;
> 2025-03-10T15:38:08.854Z) <br> &gt; OS version: Windows_NT x64 10.0.19045
> <br> &gt; Modes: <br> &gt; System Info <br> &gt; Item Value <br> &gt; CPUs
> Intel(R) Core(TM) i7-8750H CPU @ 2.20GHz (12 x 2208) <br> &gt; GPU Status
> 2d_canvas: enabled <br> &gt; canvas_oop_rasterization: enabled_on <br> &gt;
> direct_rendering_display_compositor: disabled_off_ok <br> &gt;
> gpu_compositing: enabled <br> &gt; multiple_raster_threads: enabled_on <br>
> &gt; opengl: enabled_on <br> &gt; rasterization: enabled <br> &gt;
> raw_draw: disabled_off_ok <br> &gt; skia_graphite: disabled_off <br> &gt;
> video_decode: enabled <br> &gt; video_encode: enabled <br> &gt; vulkan:
> disabled_off <br> &gt; webgl: enabled <br> &gt; webgl2: enabled <br> &gt;
> webgpu: enabled <br> &gt; webnn: disabled_off <br> &gt; Load (avg)
> undefined <br> &gt; Memory (System) 15.85GB (7.17GB free) <br> &gt; Process
> Argv C:\Users\pedra\OneDrive\Escritorio\P3_2025_03_19 <br> &gt;
> --crash-reporter-id 4cdffc02-395a-44fb-97f1-bf0f7e73a704 <br> &gt; Screen
> Reader no <br> &gt; VM 0% A/B Experiments <br> &gt; <br> &gt;
> vsliv368:30146709 <br> &gt; vspor879:30202332 <br> &gt; vspor708:30202333
> [... cut ...]
> (TerosTechnology/vscode-terosHDL#742) <br> &gt; &lt;
> &gt; <br> &gt; Type: *Bug* <br> &gt; <br> &gt; La primera línea se subraya
> en rojo y aparece el mensaje: &quot;Execution of <br> &gt; vlib.exe failed.
> this thread.Message <br> &gt; ID: ***@***.***&gt; <br> &gt; <br>
> modelsim(vlog-66)&quot;. <br> &gt; No puedo compilar el código y Code

Hola, disculpa, a veces me lio un poco con todas las extensiones y
dependencias que tiene todo. Code Runner es una extensión para ejecutar
programas en otros lenguajes, pero no debería afectar a Verilog o System
Verilog.

La cosa es que he instalado y configurado todos los paquetes según vuestra
web (
https://terostechnology.github.io/terosHDLdoc/docs/category/installation-checklist),
configurado los paths a Quartus y Modelsim y en el Verify Setup me sale
todo limpio. Sin embargo, sigue apareciendo el mismo error "Execution of
vlib.exe failed ... TerosHDL:modelsim(vlog-66)", incluso ejecutando vscode
como admin, y no sé cómo arreglarlo.

qarlosalberto, 2025-03-20:
1. Instala esta versión: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest te bajas el .vsix 
2025-03-20__11_04_teroshdl-7.0.2.vsix 
2. Ve a la tab "OUTPUT".
3. Abre "TerosHDL: Debug"
4. Abre un archivo de System Verilog
5. Guárdalo
6. Mira el comando que sale en el Debug. Será algo tipo esto:

[image]

6. Haz una caputura de pantalla y pasas las captura y el texto aquí.

Joworge, 2025-03-23:
[image]

2025-03-23 12:23:21.758 [info] Linting with command: D:\intelFPGA_lite\17.1\modelsim_ase\win32aloem\vlog.exe -quiet -nologo -sv -work C:\Users\pedra\.teroshdl_Omlc8_  "c:\QDesign_MUISE\cic_comb.sv" 
2025-03-23 12:23:21.917 [info] c:\QDesign_MUISE\cic_comb.sv || ** Error: (vlog-66) Execution of vlib.exe failed. Please check the error log for more details.
2025-03-23 12:23:21.917 [info] c:\QDesign_MUISE\cic_comb.sv || 
2025-03-23 12:23:35.854 [info] Linting with command: D:\intelFPGA_lite\17.1\modelsim_ase\win32aloem\vlog.exe -quiet -nologo -sv -work C:\Users\pedra\.teroshdl_CizfE_  "c:\QDesign_MUISE\cic_comb.sv" 
2025-03-23 12:23:35.963 [info] c:\QDesign_MUISE\cic_comb.sv || ** Error: (vlog-66) Execution of vlib.exe failed. Please check the error log for more details.
2025-03-23 12:23:35.963 [info] c:\QDesign_MUISE\cic_comb.sv || 

Mi configuración es la siguiente, por si diese alguna pista:

[image] 

[image] 

[image]

qarlosalberto, 2025-03-23:
El "Teros: Log Report" sólo es para los proyectos de Quartus.

A ver si consigo ver qué fallo tienes. Prueba estos comandos en tu cmd de Windows y ponme la salida de cada uno:

1.

```
D:\intelFPGA_lite\17.1\modelsim_ase\win32aloem\vlog.exe -quiet -nologo -sv -work C:\Users\pedra.teroshdl_Omlc8_ "c:\QDesign_MUISE\cic_comb.sv"
```

2.

```
D:\intelFPGA_lite\17.1\modelsim_ase\win32aloem\vlog.exe -quiet -nologo -sv "c:\QDesign_MUISE\cic_comb.sv"
```

3.

```
D:\intelFPGA_lite\17.1\modelsim_ase\win32aloem\vlog.exe -version
```

Joworge, 2025-03-23:
[image]

qarlosalberto, 2025-03-23:
Mira si ha generado algún archivo de log. Y mira si hay otra carpeta que no
sea esa con el binario vlog

Joworge, 2025-03-23:
No encuentro nada entre los archivos de modelsim, quartus y la carpeta del archivo .sv excepto esto, pero la fecha de modificación no es actual.

[image]

qarlosalberto, 2025-03-23:
No hay ningún log en la carpeta desde donde ejecutas el comando?

qarlosalberto, 2025-03-23:
Ejecuta antes: "vlib work" y después los otros comandos

Joworge, 2025-03-23:
No que yo vea. He buscado en:
- C:\Users\pedra
- C:\Users\pedra\.vscode\extensions\teros-technology.teroshdl-7.0.2 (y todas las subcarpetas)
- D:\intelFPGA_lite\17.1\modelsim_ase\win32aloem\  (y todas las subcarpetas)
- C:\QDesign_MUISE (no hay más archivos ni subcarpetas)

qarlosalberto, 2025-03-23:
Puede ser esto:
https://adaptivesupport.amd.com/s/question/0D52E00006iHwC1SAK/modelsim-error-vcom66-execution-of-vlib-failed?language=en_US

Joworge, 2025-03-23:
Ahora el segundo comando no da problemas, pero el primero sí:
[image]

Se me ha creado esta carpeta:
[work.zip](https://github.com/user-attachments/files/19409817/work.zip)

He ejecutado `vlib work` desde la carpeta del archivo y desde la carpeta de Teros pero el error sigue apareciendo.

qarlosalberto, 2025-03-23:
Ejecuta: vlib work
y después: D:\intelFPGA_lite\17.1\modelsim_ase\win32aloem\vlog.exe -quiet -nologo -sv "c:\QDesign_MUISE\cic_comb.sv"

qarlosalberto, 2025-03-23:
Prueba esto. Primero crea la librería en un directorio temporal y después ejecuta el parser. Si funciona lo meto en TerosHDL con un fix:

```
vlib C:\Users\pedra.teroshdl_Omlc8_
D:\intelFPGA_lite\17.1\modelsim_ase\win32aloem\vlog.exe -quiet -nologo -sv -work C:\Users\pedra.teroshdl_Omlc8_ "c:\QDesign_MUISE\cic_comb.sv"
```

closed 2025-09-22 by qarlosalberto