TerosTechnology/vscode-terosHDL #744: Lost all projects with 7.0.x version under VS code

alessio-gravina, 2025-03-20:
After installing 7.0.x version under VS code i lost all projects.

If i create a new projects if i restart VS it is lost again.

I tried to import the old .json file but it doesn't work, project list is always empty.

Reinstalled version 6.0.14 and everything works fine.

elpollodiablo84, 2025-03-20:
That happened to me also, but it's easily fixable: there are some changes in the json schema from v6 to v7, you only need to integrate them. I don't recall the entries that are changed, but if you create a dummy project, you can compare the old json with the new. I remember that now there's a new configuration entry for "vhdl-ls" or something like that, and another new entry if I'm not mistaken.

qarlosalberto, 2025-03-20:
@alessio-gravina could you try with this version?

https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/v7.0.1

alessio-gravina, 2025-03-20:
> [@alessio-gravina](https://github.com/alessio-gravina) could you try with this version?
> 
> https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/v7.0.1

I already tried but i still have the same issue.

alessio-gravina, 2025-03-20:
@qarlosalberto Thank you. The issue seems to be fixed with the version you linked. Keep up the good work.

closed 2025-03-20 by alessio-gravina