TerosTechnology/vscode-terosHDL #748: Linting with VSG sometimes lints unexpected files

JakubFranek, 2025-03-20:
**Describe the bug**
Linting with VSG sometimes unexpectedly lints VHDL files which are not interacted with in any way for unknown reason, which typically reside in TerosHDL folder. This leads to thousands of errors being reported in the Problems tab.

This seems to happen for no reason and randomly, but at least once per few hours. It happens even though I am editing only a single VHDL file and not even opening any other VHDL file at all.

See second line of log below. For some reason numeric_std.vhdl is being linted. I did not open this file ever in my VS Code, but numeric_std package is used in the file I am working on. But this happens randomly, not consistently.

```
2025-03-20 21:40:17.483 [info] Linting with command: vsg -f d:\Electronics\FPGA\Nexys-A7-Lab\source\clock_enable_generator\clock_enable_generator.vhd --all_phases -c D:\Electronics\FPGA\Nexys-A7-Lab\vsg.yaml --js C:\Users\jfran\AppData\Local\Temp\f-2025220-8000-2aqef8.e6tga 
2025-03-20 21:50:50.010 [info] Linting with command: vsg -f c:\Users\jfran\.vscode\extensions\teros-technology.teroshdl-7.0.1\server\vhdl_ls\0.83.1\vhdl_ls-x86_64-pc-windows-msvc\vhdl_libraries\ieee2008\numeric_std.vhdl --all_phases -c D:\Electronics\FPGA\Nexys-A7-Lab\vsg.yaml --js C:\Users\jfran\AppData\Local\Temp\f-2025220-8000-rn4srf.i6qzh 
2025-03-20 22:33:21.207 [info] Formatting with command: vsg  -p 2 --fix -c D:\Electronics\FPGA\Nexys-A7-Lab\vsg.yaml -f C:\Users\jfran\AppData\Local\Temp\f-2025220-8000-1ccutss.jvcf 
2025-03-20 22:33:21.238 [info] Linting with command: vsg -f d:\Electronics\FPGA\Nexys-A7-Lab\source\clock_enable_generator\clock_enable_generator.vhd --all_phases -c D:\Electronics\FPGA\Nexys-A7-Lab\vsg.yaml --js C:\Users\jfran\AppData\Local\Temp\f-2025220-8000-fxej2e.we6n 
```

Resulting Problems:

[image]

**To Reproduce**
No idea how to reproduce this consistently, but it eventually happens every evening I work within TerosHDL.

**Code**
Not relevant.

**Please complete the following information:**
 - OS: Windows
 - VSCode version 1.97.2

**TerosHDL settings**

[image]

[image]

[image]

[image]

**vsg.yaml contents**
```
rule:
    global:
        indent_size: 4
    signal_007:             # This rule requires the user to remove the default assignments.
        disable: true
    generic_007:            # This rule checks the generic names have proper case.
        case: "upper"
    port_012:               # This rule checks for default assignments on port declarations.
        disable: true
    instantiation_034:      # This rule checks for component versus direct instantiations.
        method: "entity"
    process_036:            # This rule checks for valid prefixes on process labels. The default prefix is proc_.
        disable: false
    generate_017:           # This rule checks for valid prefixes on generate statement labels. The default prefix is gen_.
        disable: false
    port_011:               # This rule checks for valid prefixes on port identifiers. The default port prefixes are: i_, o_, io_.
        disable: false
```

qarlosalberto, 2025-03-21:
Did you open that file? For example when you click in `+`, `-`, `unsigned`...

JakubFranek, 2025-03-21:
Hello @qarlosalberto,
as I mentioned in the original post, I did not open the file, not even by accident. I have no explanation for why the file is being linted.

qarlosalberto, 2025-03-21:
it's weird. If this happen again for to TerosHDL Debug Output. Now I print the linter command. Copy and paste here.

JakubFranek, 2025-03-21:
Hi @qarlosalberto,

as expected it has happened again.
I am attaching the full log from this evenings session. The unwanted file has been linted at `2025-03-21 22:57:17.141`.

[teroshdl_debug.txt](https://github.com/user-attachments/files/19398127/teroshdl_debug.txt)

qarlosalberto, 2025-03-21:
Ok, I will take a look when I have time.