TerosTechnology/vscode-terosHDL #751: Global config manager doesn't parse project list at extension start after update to v7.0 [solved?]

irshbn, 2025-03-23:
**Describe the bug**
After updating from TerosHDL v6.0.14 to v7.0.1, when extension starts, global config manager tries to initiate a multi project manager which throws a ```ProjectOperationError``` when loading the existing ```.teroshdl2_prj.json``` . Loading a global configuration works fine, it is only the project list that doesn't get parsed correctly.  New projects created from within extension interface overwrite the existing _prj file, but after restarting the extension newly created projects do not get parsed as well. 

**To Reproduce**
1. Run VSCode with TerosHDL v6.0.14 installed
2. Create dummy project from within interface
3. Check that .teroshdl2_prj.json exists in user home dir and the dummy project config is written to it
4. Update to TerosHDL v7.0.1 via VSCode extension marketplace
5. Initiate extension to see the project list empty and a warning in debug logs

**Logs**
(sic)
```
nsion "TerosHDL" is now active!
2025-03-23 09:23:07.481 [warning] There have been errors loading project list from disk.
2025-03-23 09:23:07.498 [info] activated language provider
2025-03-23 09:23:07.498 [info] activated template manager
2025-03-23 09:23:07.498 [info] activated documenter
2025-03-23 09:23:07.498 [info] activated state machine
2025-03-23 09:23:07.498 [info] activated schematic
2025-03-23 09:23:07.498 [info] activated linter
2025-03-23 09:23:07.498 [info] activated formatter
2025-03-23 09:23:07.498 [info] activated completions
2025-03-23 09:23:07.498 [info] activated shutter mode
2025-03-23 09:23:07.498 [info] activated config viewer
2025-03-23 09:23:07.498 [info] activated views
2025-03-23 09:23:07.498 [info] activated comander
```

**Proposed solution**
I was able to solve the problem by backing up my projects in EDAM yaml files, then deleting the _config and _prj files and reinstalling the extension after that. After reinstallation the extension creates new _config and _prj which get parsed correctly and the bug vanishes. All projects added to multi project manager via loading from EDAM file are saved normally to _prj file and the correct workflow is restored.

**Please complete the following information:**
 - OS: Ubuntu 24.04 via WSL2
 - VSCode version 1.98.2
 - TerosHDL version 6.0.14 -> 7.0.1

**Additional context**
I assume that it could be a bug not related to TerosHDL per se, and rather some bug with VSCode running remotely in WSL, or with WSL file system itself, or with a random read access right violation, or something else. If this bug cannot be reproduced, feel free to close the issue as unrelated.

qarlosalberto, 2025-03-23:
Can you try this version? I will do the release today 

https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest

SAHassani, 2025-03-25:
Indeed version 7.0.3 loads the projects at startup. Thank you very much for resolving the issue!  :-)

closed 2025-03-25 by qarlosalberto