TerosTechnology/vscode-terosHDL #752: "Module 'x' not found!" when opening state machine viewer (VHDL, Windows)

Daellhin, 2025-03-23:
**Describe the bug**
Schematic viewer does not work for VHDL files and gives error 'ERROR: Module `half_adder' not found!' (see screenshot)

**To Reproduce**
Open GHDL example project, open `half_adder.vhd` file, open schematic viewer.

**Please complete the following information:**
 - OS: Windows 11 24H2
 - VSCode version: 1.98.2
 - TerosHDL version: 7.0.1

**Screenshots**
[image]

**Additional context**
Setup checks all pass, except modelsim (but is unused)

qarlosalberto, 2025-03-24:
It's weird. It seems that you are using yowasp-yosys instead Yosys with GHDL. Take a look to the command output.

Daellhin, 2025-03-24:
Oops, I forgot to apply the changes when selecting the backend. Got a bit to used to other dialogs in vscode that save on edit.

closed 2025-03-24 by Daellhin