TerosTechnology/vscode-terosHDL #755: Merge vsg linter and formatter settings

sigmuha, 2025-03-27:
**Is your feature request related to a problem? Please describe.**
It is confusing to have to set the settings for vsg two times: One for the editor linter, and one for the formatter. If you haven't configured them both the same, it is confusing that when you hit the format button, it doesn't format according to the errors given by the linter.

**Describe the solution you'd like**
Merge the settings under Formatter --> VHDL VSG, and Linter Settings --> VSG linter. I don't see a situation where you would want these to be different from each other.

JennySmith888, 2025-03-27:
+1 to this I am trying to set up Teros HDL with VSG for the first time and I'm so confused how to configure the two together...

Running into weird errors during setup verification like it finds vsg in my python venv
```
⊙ Checking Python dependencies. Current configured installation path: "/home/jpsmith/python_venvs/generic_fpga/bin/python3"
  🎉 vunit found.
  🎉 edalize found.
  🎉 cocotb (optional) found.
  🎉 vsg (optional) found.
```
But the same check fails during the Formatter configuration check??
```⊙ Checking formatter vsg for VHDL.
  🔎 Searching for the binary "Vsg" in the system path
  ❌ Vsg not found in the system path. Search executed with: "vsg --version"
👎 The linter installation path is not correctly configured. The binary could not be located in either the system path or the configured path. Please check the documentation.```

qarlosalberto, 2025-03-28:
@JennySmith888 does it work for you VSG? In theory both should fail.

1. Do an import in python: import vsg
2. Do "vsg --version" in the command line

Could you try manually in your terminal?

sigmuha, 2025-04-03:
It looks like this has been updated in #756. Can you confirm @qarlosalberto?

sigmuha, 2025-04-04:
I've tried 
Additionally, I tried this in pre-release `2025-03-28__09_45_teroshdl-7.0.5.vsix`., and i seem to have a new issue. 
In a file I get this warning:
[image]
My vsg-config looks like this:
[image]

In my one of the vsg-settings-files I have configured the global indentation for be 4 indents, like:
```yaml
rule:
  global:
    indent_size: 4
```

I don't have any references to `attribute_declaration_300` anywhere in my configuration files. Also when running vsg manually, and copy-pasting the paths under `Path JSON or YAML configuration file.`, I get other, correct warning according to the configuration files.

qarlosalberto, 2025-04-04:
I have moved the configuration. Did you configure it correctly?

sigmuha, 2025-04-04:
In the pre-release version (7.0.5) I configured it exactly like in the previous versions, with the exact same paths. Different warnings were given in 7.0.5 (like the screenshot), than in 7.0.3 (which are correct according to the vsg configuration). The one error i showed was just one of many erroneous errors.

qarlosalberto, 2025-04-04:
Use: https://github.com/TerosTechnology/vscode-terosHDL/releases/tag/latest
And let me know the issues.

sigmuha, 2025-04-04:
As I have stated, this is the version which gave me the issues shown. I installed `2025-03-28__09_45_teroshdl-7.0.5.vsix`, and the issues with the mismatch in vsg-warnings happened then. When I downgraded to the latest stable release, this issue disappeared.

qarlosalberto, 2025-04-04:
But what is your configuration. As I said I have moved the VSG configuration. You need to reconfigure VSG:

[image]

sigmuha, 2025-04-04:
The image in https://github.com/TerosTechnology/vscode-terosHDL/issues/755#issuecomment-2777848152 shows my configuration.  The exact same paths were used in this as in the the one that worked.

qarlosalberto, 2025-04-04:
Show me the debug output. You will see VSG command: Output -> TerosHDL: Debug

sigmuha, 2025-04-04:
I have to anonymize some paths, but this is the output:
```
nsion "TerosHDL" is now active!
2025-04-04 10:39:19.449 [info] activated language provider
2025-04-04 10:39:19.449 [info] activated template manager
2025-04-04 10:39:19.449 [info] activated documenter
2025-04-04 10:39:19.449 [info] activated state machine
2025-04-04 10:39:19.449 [info] activated schematic
2025-04-04 10:39:19.479 [info] activated linter
2025-04-04 10:39:19.480 [info] activated formatter
2025-04-04 10:39:19.480 [info] activated completions
2025-04-04 10:39:19.480 [info] activated shutter mode
2025-04-04 10:39:19.480 [info] activated config viewer
2025-04-04 10:39:19.539 [info] activated views
2025-04-04 10:39:19.539 [info] activated comander
2025-04-04 10:39:19.539 [info] Linting with command: vsg -f c:\Users\sigmuh\projects\test\top.vhd --all_phases --js C:\Users\sigmuh\AppData\Local\Temp\f-202534-37760-x5uhv8.yi98n 
```
It seems like the configurations are not used, and what I'm seeing in the editor are the default ones set by vsg.

qarlosalberto, 2025-04-04:
I don't know why it happens... In debugging it in my PC and it works fine. 

It seems that your path is empty but I don't know why... Try to uninstall and install TerosHDL again.

sigmuha, 2025-04-04:
Are the paths sanitized differently than in the previous version? How are the paths expected to look like? How do your paths look like?