TerosTechnology/vscode-terosHDL #759: Reduce file pollution in home directory on windows

sigmuha, 2025-04-04:
**Is your feature request related to a problem? Please describe.**
The home directory of the user that has installed terosHDL gets a very polluted home directory from terosHDL generating many, seemingly empty, and unused files. Also, some other files that are used (like `.vhdl_ls.toml`, `.teroshdl2_config.json`) should be considered moved, as the user should not edit these files manually. 

**Describe the solution you'd like**
Remove the unused files, like:
[image]

Move the active files to a directory more appropriate for a vs-code extention. Examples of such files:
[image]

**Additional context**
OS: Windows 11

qarlosalberto, 2025-04-04:
What is the content of .teroshdl_*** with ***  equal to random?

qarlosalberto, 2025-04-04:
Is the latest version generating them? Or are they old?

sigmuha, 2025-04-04:
No, not the latest version (yet). They originate from various dates, including yesterdat, when I used version 7.0.3. I'm don't know where these files originate from

qarlosalberto, 2025-04-04:
I did some fixes related to that. But maybe I miss something. If this happen again try to remember in which feature.

sigmuha, 2025-04-04:
From #760 we can see that the empty garbage files are created as a part of the schematic viewer.

qarlosalberto, 2025-04-04:
What version are you using?

sigmuha, 2025-04-04:
I see. I'm using 7.0.3. Closing as this is fixed in #413