TerosTechnology/vscode-terosHDL #760: Schematic viewer: cannot configure on windows.

sigmuha, 2025-04-04:
**Describe the bug**
When configuring the schematic viewer for vhdl (GHDL + Yosys) and adding a path to start the oss-cad-suite environment, the `TerosHDL: Global` terminal doesn't produce any output, and the schematic viewer doesn't work. 

The command used to activate the environment is: `C:\oss-cad-suite\start.bat &`, which works fine (eg. `yosys`is found) in a CMD-shell. This command is run, as per the suggestion in the field description above: *Extra options passed before to run yowasp-yosys or yosys. Eg: 'conda activate base & ' or 'C:/OSS-CAD-SUITE/oss-cad-suite/start.bat & '*. This might be related to starting a background process with `&`, and the terminal output is piped somewhere else?

When removing said command, output is again produced, but initializing the schematic viewer naturally fails:
```
*************************************************
Checking Schematic configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/schematic_viewer/installation) 
❌ Some checks failed ❌
*************************************************
⊙ Selected backend: yosys_ghdl with installation path: "". 
  🔎 Searching for the binary "yosys_ghdl" in the system path
  ❌ yosys_ghdl not found in the system path. Search executed with: "yosys --version"
👎 The Schematic Backend installation path is not correctly configured. The binary could not be located in either the system path or the configured path. Please check the documentation.
```
This issue also doesn't occur when having ``C:\oss-cad-suite\environment.bat` in the same extra options field. However, the schematic viewer still isn't configured correctly:
```
*************************************************
Checking Schematic configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/schematic_viewer/installation) 
❌ Some checks failed ❌
*************************************************
⊙ Selected backend: yosys_ghdl with installation path: "". 
  🔎 Searching for the binary yosys_ghdl at "C:\oss-cad-suite\environment.bat"
  ✅ yosys_ghdl found at "C:\oss-cad-suite\environment.bat" using the command: "C:\oss-cad-suite\environment.bat --version"
🔧 Schematic Backend was found; however, the installation path is not correctly configured. Ensure the path points to the binary folder, not the specific binary. Found in: "C:\oss-cad-suite\environment.bat"
```

How do I configure the schematic viewer correctly? What is meant to be passed in the "extra options" field when using oss-cad-suite?

**Please complete the following information:**
 - OS: Windows 11
 - Extention version: 7.0.3

qarlosalberto, 2025-04-04:
It seems that `oss-cad-suite` doesn't support GHDL plugin for Yosys. Try this: https://github.com/ghdl/ghdl-yosys-plugin?tab=readme-ov-file#pre-built-packages

sigmuha, 2025-04-04:
Even though this isn't supported, these issues (as I see it) are related to not being able to see terminal output when using the recommended command `.../start.bat &`. This comment needs to be considered changed.

It also seems as the `.../environment.bat` command works better, where it seems to find "yosys_ghdl". However the command "C:\oss-cad-suite\environment.bat --version" seems strange, so I don't know the validity on that.