TerosTechnology/vscode-terosHDL #770: Verify Setup does not find yosys

glembler, 2025-05-06:
**Describe the bug**
A clear and concise description of what the bug is.

Verify Setup command does not find yosys

**To Reproduce**
Code to reproduce the error.

Press "Verify Setup" button in the TerosHDL interface.

**Code**
Code to reproduce the error.

(NA)

**Please complete the following information:**
 - OS: [e.g.Ubuntu 18.04]
Ubuntu 22.04.5 LTS

 - VSCode version version [e.g. 22]
Version: 1.99.3
Commit: 17baf841131aa23349f217ca7c570c76ee87b957
Date: 2025-04-15T23:18:46.076Z
Electron: 34.3.2
ElectronBuildId: 11161073
Chromium: 132.0.6834.210
Node.js: 20.18.3
V8: 13.2.152.41-electron.0
OS: Linux x64 6.8.0-58-generic snap

**Screenshots**
If applicable, add screenshots to help explain your problem. **Share the code as text, not as a screenshots!**

*************************************************
Checking Schematic configuration (https://terostechnology.github.io/terosHDLdoc/docs/guides/schematic_viewer/installation) 
❌ Some checks failed ❌
*************************************************
⊙ Selected backend: yosys with installation path: "/usr/bin". This backend doesn't support VHDL files.
  🔎 Searching for the binary yosys at "/usr/bin/yosys"
  ❌ yosys not found at "/usr/bin/yosys". Search executed with: "/usr/bin/yosys --version"
  🔎 Searching for the binary yosys at "/usr/bin/yosys.exe"
  ❌ yosys not found at "/usr/bin/yosys.exe". Search executed with: "/usr/bin/yosys.exe --version"
  🔎 Searching for the binary yosys at "/usr/bin"
  ❌ yosys not found at "/usr/bin". Search executed with: "/usr/bin --version"
  🔎 Searching for the binary yosys at "/usr/bin.exe"
  ❌ yosys not found at "/usr/bin.exe". Search executed with: "/usr/bin.exe --version"
  🔎 Searching for the binary "yosys" in the system path
  ❌ yosys not found in the system path. Search executed with: "yosys --version"
👎 The Schematic Backend installation path is not correctly configured. The binary could not be located in either the system path or the configured path. Please check the documentation.

**Additional context**
Add any other context about the problem here.

yosys is installed in a system path and working

$ which yosys
/usr/bin/yosys

$ yosys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2019  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.9 (git sha1 1979e0b)

yosys> help

    abc                  use ABC for technology mapping
    add                  add objects to the design
    ...

glembler, 2025-05-06:
TerosHDL checks yosys version with "--version" option. yosys option "--version" is invalid.

$ /usr/bin/yosys --version
/usr/bin/yosys: invalid option -- '-'
Run '/usr/bin/yosys -h' for help.

qarlosalberto, 2025-09-22:
Try a newer version of Yosys. It working for me with:

```
 /----------------------------------------------------------------------------\
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |  Copyright (C) 2012 - 2025  Claire Xenia Wolf <claire@yosyshq.com>         |
 |  Distributed under an ISC-like license, type "license" to see terms        |
 \----------------------------------------------------------------------------/
 Yosys 0.49+10 (git sha1 18a7c0038, clang++ 18.1.8 -fPIC -O3)

```

beautyseeker, 2026-02-02:
it seems terosHDL not compatible with latedst oss-cad-suit:
terosHDL version:7.0.3
IDE:VS code 1.108.1
bash CLI output:

$ uname -a  
Linux JeffreysENIAC 6.14.0-37-generic #37~24.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Nov 20 10:25:38 UTC 2 x86_64 x86_64 x86_64 GNU/Linux  
$ /projects/ysyx_practice$ which yosys
/home/passion/Downloads/oss-cad-suite/bin/yosys
$ /projects/ysyx_practice$ yosys --version
Yosys 0.61+44 (git sha1 57ac113b7, clang++ 18.1.8 -fPIC -O3)