TerosTechnology/vscode-terosHDL #787: Command 'TerosHDL: Verify Setup' resulted in an error

WuShFeng, 2025-08-04:
> command 'teroshdl.configuration.verifySetup' not found
[image]

qarlosalberto, 2025-08-04:
What TerosHDL version are you using?

qarlosalberto, 2025-09-22:
Feel free to reopen.

gmartina, 2025-11-08:
@qarlosalberto 
I could not reproduce the error with:

TerosHDL version: 7.0.3
Windows 11 Pro 25H2
Version: 1.105.1

HuskyNator, 2025-11-08:
This happens for me both when i use the command pallete as well as the GUI button.
Saving my global settings does succesfully confirm my setup however (from what i can tell).

evanwporter, 2025-11-14:
@gmartina 

So maybe this is a different issue, but my global logs don't show up:

[image]

The debug ones do though!

[image]

closed 2025-12-10 by qarlosalberto